ophub/amlogic-s9xxx-openwrt #418: X96Max_Plus_A100 issue

ashkib, 2023-04-13:
**Device Information
 - SOC: S095x3
 - Model [X96Max_Plus_A100] 
 - 4GB RAM and 32GB emmc 
 - WIFI AM7256
 - LAN 100M
 
Openwrt_official_amlogic_s905x3_k5.15.106_2023.04.11.img.gz

**Describe the bug 
Dear Sir, thank you for your effort and support
I have X96Max_Plus_A100 box, It works fine with USB boot both wifi and lan works but after emmc install I always have Black screen no log and no Ethernet IP
Could you please help me to install Openwrt on emmc ?
Kind regards

Board photo 
[https://photos.app.goo.gl/2yGYDmMtH42WBSK66](url)

ophub, 2023-04-14:
```
armbian-install -m yes
523
```

ashkib, 2023-04-14:
Thank you ophub for your answer
Could you please help me by explaining more details about the steps ?

ophub, 2023-04-14:
I answered several warehouse questions at the same time, but I made a mistake. You are using the OP system, and the commands above are from Armbian.

In the menu of the openwrt system, select System > amlogic service > Plugin settings > Use `Bootloader`, and then install the OP to use the mainline u-boot

ashkib, 2023-04-14:
thank you for your support
unfortunately still not booting from emmc after checking write boot loader
openwrt_official_amlogic_s905x3_k5.15.106_2023.04.13.img.gz

ophub, 2023-04-14:
https://github.com/ophub/amlogic-s9xxx-armbian/issues/779

This is a previous test post, openwrt and armbian system startup requirements are the same.

ophub, 2023-04-14:
https://github.com/ophub/kernel/releases/tag/tools
a100

The network card cannot be recognized, you may need to replace the Android system first, and then flash armbian

ashkib, 2023-04-14:
I flashed the android rom from the link 
used the dtb from the link above (Q1) to boot from usb , both lan and wifi works
activated boot loader writing and installed openwrt on emmc using (device 515 ) in the list
emmc success but with 2 GB ram only of 4GB

Thank you for your help 
You can close the issue now

ophub, 2023-04-15:
https://github.com/ophub/amlogic-s9xxx-armbian/tree/main/build-armbian/documents

look at: 12.12
copy u-boot

ashkib, 2023-04-15:
ok thank you so much ..
I will try it and update you. 

Could you please recommend the most compatible dtb file for my tv box for emmc install?
https://photos.app.goo.gl/2yGYDmMtH42WBSK66

SOC: S095x3
Model X96Max_Plus_A100
4GB RAM and 32GB emmc
WIFI AM7256
LAN 100M

ophub, 2023-04-15:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/cf85b0f4ba79acb5ea89746566a81c2e919b52eb/make-openwrt/openwrt-files/common-files/etc/model_database.conf#L153

523

ashkib, 2023-04-15:
523 boots and works from USB but gave me black screen after emmc install 
Could you guide me to the right steps to install openwrt using 523 dtb?

ophub, 2023-04-15:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/make-openwrt/documents

ashkib, 2023-04-15:
Could you please be more specific
example easy steps to follow 1,2,3...

closed 2023-04-19 by ophub

hillz2, 2023-04-26:
> Could you please be more specific example easy steps to follow 1,2,3...

I also have X96 Max Plus running openwrt but the LAN port doesn't work, does yours work ?