ophub/amlogic-s9xxx-openwrt #421: 关于mgv2000的openwrt固件无法安装某些ipk的问题

timeout-error, 2023-04-28:
大佬你好，很抱歉打扰您，我的盒子型号是MGV2000（JL)，配置是s905l2，1+8.刷的openwrt包是openwrt_lede_amlogic_s905l2_k5.15.108_2023.04.25.img，K5跟K6的版本都试过了，目前只有K5版本可以启动，K6的话能显示一半的代码，然后显示倒数三秒就不断重启，无法进入系统，只有k5版本能正常使用。目前遇到的问题是我想安装openclash插件当旁路由使用，结果自带的软件源无法安装部分ipk，比如ruby，kmod-inet-diag等。因为我个人比较小白，想要安装openclash，不会自己编译固件只能跟着openclash的官方文档慢慢走，自带的腾讯源是搜不到ruby的，更换中科大源后成功安装ruby，然后也能搜到kmod-inet-diag，但是安装kmod（这里简写为kmod）就会报错,说是源与内核不匹配Collected errors: * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.176-1-9ba0cc12a8e6bbf68463d91f0091bf15) for kmod-inet-diag
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-inet-diag found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-inet-diag.问了懂一些的朋友说是我的源有问题，换了清华源跟openwrt官方源也是回报错，实在是不知道怎么解决了，希望大佬能教教我该怎么操作才能安装openclash

ophub, 2023-04-28:
kmod无法安装，需要在编译时集成。第三方插件的集成方法都一样，可以看文档里的说明，也可以根据插件网站的介绍集成。

closed 2023-04-29 by ophub

timeout-error, 2023-04-29:
好吧，多谢大佬解答