ophub/amlogic-s9xxx-openwrt #423: CM311-1 CH版本，写入EMMC后无法启动

Becods, 2023-05-01:
**Device Information | 设备信息**
 - SOC: s905l3b
 - Model CH311-1 CH

**Armbian Version | 系统版本**
 - Kernel Version: 5.15.109
 - Release: official_04.27.0623

**Describe the bug | 问题描述**
使用从release中下载的固件

解压后文件名`openwrt_official_amlogic_s905l3b_k5.15.109_2023.04.27.img`

写入EMMC后无法启动，U盘启动正常

box model选择112，在armbian中测试过能正常通过EMMC启动

[U盘启动uboot日志](https://pastebin.com/4HiHnXmw)

[EMMC启动uboot日志](https://pastebin.com/Uk0AcJiT)

[U盘启动系统日志](https://pastebin.com/Tfi26QfN)

<details><summary>安装日志</summary>
<p>

```
root@OpenWrt:/# openwrt-install-amlogic
AUTO_MAINLINE_UBOOT: 
SPECIFY_DTB: 
SPECIFY_SOC: 
SPECIFY_UBOOT: 
SHARED_FSTYPE: 
Current device: Amlogic Meson GXL (S905L3b) MBH-M302A Box [ amlogic ]
/dev/mmcblk2
mmcblk2 : 15634268160 bytes
ROOTFS: sda2
BOOT: sda1
Please select s9xxx box model:
----------------------------------------------------------------------------------------------------
 101. Phicomm-N1                    s905d   2GB-Mem,1Gb-Nic,brcm43455-wifi
 102. Phicomm-N1(DMA-thresh)        s905d   When-ethmac-flow-control-is-off
 103. HG680P                        s905x   2+8G,100Mb-Nic
 104. TX3-Mini                      s905w   2+8G,100Mb-Nic
 105. MECOOL-KI-Pro                 s905d   2G/16G,1Gb-Nic
 106. T95                           s905x   2G/16G,1Gb-Nic
 107. B860H                         s905x   2+8G,100Mb-Nic
 108. TBee                          s905x   2+8G,100Mb-Nic
 109. W95                           s905w   2GB-RAM,16GB-ROM,100Mb-Nic
 110. X96-Mini                      s905w   2+8G,100Mb-Nic
 111. TX9                           s905x   100Mb-Nic,rtl8723cs-WLAN
 112. M302A,M304A,CM211-1,CM311-1   s905l3b 2+8G,100Mb-Nic
 113. MGV2000,MGV3000,M301A         s905l2  2+8G,100Mb-Nic
 114. Wojia-TV-IPBS9505             s905l2  2+8G,100Mb-Nic
 115. Q96-mini                      s905lb  1G/8G,100Mb-Nic
 116. CM311-1,HG680-LC,M401A        s905l3  2+8G,100Mb-Nic
 117. E900V22E                      s905l3b 2+8G,100Mb-Nic
 118. BesTV-R3300L,SumaVision-Q7    s905lb  2+8G,100Mb-Nic
 119. X96W,FunTV,MXQ-Pro-4K         s905w   2GB-RAM,16GB-ROM,100Mb-Nic,AP6255-wifi/BT
 120. E900V22D                      s905l3b 2+8G,100Mb-Nic
 121. UNT400G1                      s905l3  2+8G,100Mb-Nic
 122. E900v21E                      s905l2  2+8G,100Mb-Nic
 123. IP103H,TY1608                 s905l3b 2+8G,100Mb-Nic
 124. CM211-1                       s905l3  2+8G,100Mb-Nic
 201. Octopus-Planet                s912    2GB-Mem,1Gb-Nic
 202. Octopus-Planet(FAKE)          s912    2GB-Mem,1Gb-Nic
 203. H96-Pro-Plus                  s912    2G/32G,1Gb-Nic
 204. Tanix-TX92                    s912    3GB-DDR4,32GB-eMMC,5G-WIFI,1Gb-Nic
 205. VORKE-Z6-Plus                 s912    3GB-DDR3,32GB-eMMC,1Gb-Nic
 206. MECOOL-M8S-Pro-L              s912    2G/3G-RAM,16G/32G-ROM,100Mb-Nic
[... 563 lines cut ...]
							      >		write uboot to the boot device
							      >	store erase boot/data: 
							      >		erase the area which is uboot or data 
							      >	store erase partition <partition_name>: 
							      >		erase the area which partition in u-boot 
							      >	store erase dtb 
							      >	store erase key 
							      >	store disprotect key 
							      >	store rom_protect on/off 
							      >	store scrub off|partition size
							      >		scrub the area from offset and size 
							      >	store dtb iread/read/write addr <size>
							      >		read/write dtb, size is optional 
							      >	store key read/write addr <size>
							      >		read/write key, size is optional 
							      >	store mbr addr
							      >	   update mbr/partition table by dtb

uboot time: 37380823 us					      \	cmd store failed 
							      >	Err imgread(L135):Fail to read 0x100000B from part[recovery] 
							      >	gxl_p211_v1#
```

</p>
</details>

ophub, 2023-05-01:
大部分魔百和都要先把当前的安卓4的系统刷成有root权限安卓9系统，才能正常刷armbian/openwrt系统。具体情况从首页的设备列表点进去看对应设备的测试报告。

安卓系统下载备份：
https://github.com/ophub/kernel/releases/tag/tools

Becods, 2023-05-01:
> 大部分魔百和都要先把当前的安卓4的系统刷成有root权限安卓9系统，才能正常刷armbian/openwrt系统。具体情况从首页的设备列表点进去看对应设备的测试报告。
> 
> 安卓系统下载备份： https://github.com/ophub/kernel/releases/tag/tools

已经刷过9.0了，并且armbian是能够正常安装并从emmc启动的

[image]

<details><summary>安装日志</summary>
<p>

```
root@armbian:~# armbian-install
[ STEPS ] Start install Armbian to eMMC...
[ STEPS ] Start checking dependencies...
[ INFO ] Dependency check completes, Start installing Armbian.
[ STEPS ] Start initializing the environment...
[ INFO ] Use mainline u-boot: [ no ]
[ INFO ] Using the Ampart tool: [ yes ]
[ INFO ] Show all lists: [ no ]
[ INFO ] The device eMMC name: [ /dev/mmcblk2 ]
[ STEPS ] Start selecting device...
--------------------------------------------------------------------------------------
ID    SOC        MODEL                          DTB                                               
--------------------------------------------------------------------------------------
101   s905d      Phicomm-N1                     meson-gxl-s905d-phicomm-n1.dtb                    
102   s905d      Phicomm-N1(DMA-thresh)         meson-gxl-s905d-phicomm-n1-thresh.dtb             
103   s905x      HG680P                         meson-gxl-s905x-p212.dtb                          
104   s905w      TX3-Mini                       meson-gxl-s905w-tx3-mini.dtb                      
105   s905d      MECOOL-KI-Pro                  meson-gxl-s905d-mecool-ki-pro.dtb                 
106   s905x      T95                            meson-gxl-s905x-p212.dtb                          
107   s905x      B860H                          meson-gxl-s905x-b860h.dtb                         
108   s905x      TBee                           meson-gxl-s905x-tbee.dtb                          
109   s905w      W95                            meson-gxl-s905w-p281.dtb                          
110   s905w      X96-Mini                       meson-gxl-s905w-x96-mini.dtb                      
--------------------------------------------------------------------------------------
111   s905x      TX9                            meson-gxl-s905x-tx9.dtb                           
112   s905l3b    M302A,M304A,CM211-1,CM311-1    meson-gxl-s905l3b-m302a.dtb                       
113   s905l2     MGV2000,MGV3000,M301A          meson-gxl-s905l2-x7-5g.dtb                        
114   s905l2     Wojia-TV-IPBS9505              meson-gxl-s905l2-ipbs9505.dtb                     
115   s905lb     Q96-mini                       meson-gxl-s905x-p212.dtb                          
116   s905l3     CM311-1,HG680-LC,M401A         meson-gxl-s905l2-x7-5g.dtb                        
117   s905l3b    E900V22E                       meson-gxl-s905l3b-e900v22e.dtb                    
118   s905lb     BesTV-R3300L,SumaVision-Q7     meson-gxl-s905x-p212.dtb                          
119   s905w      X96W,FunTV,MXQ-Pro-4K          meson-gxl-s905w-x96w.dtb                          
120   s905l3b    E900V22D                       meson-gxl-s905l2-x7-5g.dtb                        
--------------------------------------------------------------------------------------
121   s905l3     UNT400G1                       meson-gxl-s905l2-x7-5g.dtb                        
122   s905l2     E900v21E                       meson-gxl-s905l2-x7-5g.dtb                        
123   s905l3b    IP103H,TY1608                  meson-gxl-s905l3b-m302a.dtb                       
124   s905l3     CM211-1                        meson-gxl-s905l3b-m302a.dtb                       
0     Other      Customize                      Enter-custom-dtb-name                             
--------------------------------------------------------------------------------------
[ OPTIONS ] Please Input ID: 112
[ INFO ] Input Box ID: [ 112 ]
[ INFO ] FDTFILE: [ meson-gxl-s905l3b-m302a.dtb ]
[ INFO ] MAINLINE_UBOOT: [  ]
[ INFO ] BOOTLOADER_IMG:  [  ]
[ INFO ] UBOOT_OVERLOAD: [ u-boot-s905x-s912.bin ]
[ INFO ] NEED_OVERLOAD: [ no ]
[ STEPS ] Start selecting file system type...
-----------------------------------------------
  ID   TYPE
-----------------------------------------------
  1    ext4
  2    btrfs
-----------------------------------------------
[ OPTIONS ] Please Input ID (1/2): 2
[ INFO ] Input Type ID: [ 2 ]
[ INFO ] The type of file system: [ btrfs ]
[... 900 lines cut ...]
[   18.070322] meson8b-dwmac c9410000.ethernet eth0: PHY [0.e40908ff:08] driver [Meson GXL Internal PHY] (irq=49)
[   19.610592] meson8b-dwmac c9410000.ethernet eth0: No Safety Features support found
[   19.612558] meson8b-dwmac c9410000.ethernet eth0: PTP not supported by HW
[   19.620435] meson8b-dwmac c9410000.ethernet eth0: configuring for phy/rmii link mode
[  OK  ] Finished Network Manager Wait Online.
[  OK  ] Reached target Network is Online.
         Starting /etc/rc.local Compatibility...
[  OK  ] Closed OpenBSD Secure Shell server socket.
         Starting OpenBSD Secure Shell server...
[  OK  ] Started OpenBSD Secure Shell server.
[   30.950940] rc.local[1360]: Restarting ssh (via systemctl): ssh.service.
[   31.013001] rc.local[1373]: irq name:eth0, irq:29, affinity: 8
[   31.013503] rc.local[1373]: irq name:xhci-hcd:usb1, irq:48, affinity: 4
[   31.013965] rc.local[1373]: Set the rps cpu mask of eth0 to 0x3
[  OK  ] Started /etc/rc.local Compatibility.
         Starting Hold until boot process finishes up...
         Starting Terminate Plymouth Boot Screen...

Armbian 23.05.0-trunk Bullseye ttyAML0 

armbian login:
```

</p>
</details>

ophub, 2023-05-01:
armbian-install -m yes

Becods, 2023-05-02:
> armbian-install -m yes

能够启动armbian，但无法启动openwrt

[armbian emmc启动日志](https://pastebin.com/fH0MTf2i)

[armbian安装日志](https://pastebin.com/Lp4xd5V1)

[openwrt u盘启动日志](https://pastebin.com/q5c5JCV9)

[openwrt安装日志](https://pastebin.com/REBad0CE)

[openwrt emmc启动日志](https://pastebin.com/5YNjJuqU)

<details><summary>armbian与openwrt从emmc启动的uboot日志对比</summary>
<p>

```
GXLX2:BL1:3cfee7:42a5ae;FEAT:ADFD318C:0;POC:3;RCY:0;EMMC:0;RE	GXLX2:BL1:3cfee7:42a5ae;FEAT:ADFD318C:0;POC:3;RCY:0;EMMC:0;RE
TE: 275623						      |	TE: 144230
 								 
BL2 Built : 13:35:42, Jun  1 2021. 				BL2 Built : 13:35:42, Jun  1 2021. 
gxl g49d5acc - liqiang.hou@droid07-sz				gxl g49d5acc - liqiang.hou@droid07-sz
 								 
Board ID = 3, adc=225						Board ID = 3, adc=225
set vcck to 1070 mv						set vcck to 1070 mv
set vddee to 1070 mv						set vddee to 1070 mv
CPU clk: 1200MHz						CPU clk: 1200MHz
 								 
DDR4 chl: Rank0+1 @ 912MHz - FAIL				DDR4 chl: Rank0+1 @ 912MHz - FAIL
DDR4 chl: Rank0 @ 912MHz					DDR4 chl: Rank0 @ 912MHz
bist_test rank: 0 1e 08 35 32 1a 4a 18 00 31 34 1a 4f 12 00 2 |	bist_test rank: 0 1f 08 36 32 1a 4a 18 00 31 34 1a 4f 13 00 2
 								 
Rank0: 2048MB(auto)-2T-18					Rank0: 2048MB(auto)-2T-18
AddrBus test pass!						AddrBus test pass!
-s								-s
Load fip header from eMMC, src: 0x0000c200, des: 0x01400000, 	Load fip header from eMMC, src: 0x0000c200, des: 0x01400000, 
New fip structure!						New fip structure!
Load bl30 from eMMC, src: 0x00010200, des: 0x013c0000, size: 	Load bl30 from eMMC, src: 0x00010200, des: 0x013c0000, size: 
Load bl301 from eMMC, src: 0x00018200, des: 0x01380000, size:	Load bl301 from eMMC, src: 0x00018200, des: 0x01380000, size:
Load bl31 from eMMC, src: 0x0001c200, des: 0x10100000, size: 	Load bl31 from eMMC, src: 0x0001c200, des: 0x10100000, size: 
Load bl33 from eMMC, src: 0x00038200, des: 0x01000000, size: 	Load bl33 from eMMC, src: 0x00038200, des: 0x01000000, size: 
NOTICE:  BL3-1: v1.0(debug):27617c8a4				NOTICE:  BL3-1: v1.0(debug):27617c8a4
NOTICE:  BL3-1: Built : 18:02:23, Aug 18 2021			NOTICE:  BL3-1: Built : 18:02:23, Aug 18 2021
NOTICE:  BL31: GXL normal boot!					NOTICE:  BL31: GXL normal boot!
NOTICE:  BL31: BL33 decompress pass				NOTICE:  BL31: BL33 decompress pass
[Image: gxl_v1.1.3103-923400455-dirty 2020-12-28 18:42:52 xin	[Image: gxl_v1.1.3103-923400455-dirty 2020-12-28 18:42:52 xin
efuse init ops = c5						efuse init ops = c5
efuse init hdcp = c, cf9=7					efuse init hdcp = c, cf9=7
x2_hp_e = 0							x2_hp_e = 0
bl30: check_permit, count is 1					bl30: check_permit, count is 1
bl30: check_permit: ok!						bl30: check_permit: ok!
chipid: 0 0 3 c c 0 f1 b3 40 a0 0 c5 not ES chip		chipid: 0 0 3 c c 0 f1 b3 40 a0 0 c5 not ES chip
[0.818543 Inits done]					      |	[0.685692 Inits done]
INFO:    BL3-1: Initializing runtime services			INFO:    BL3-1: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader			WARNING: No OPTEE provided by BL2 boot loader
ERROR:   Error initializing runtime service opteed_fast		ERROR:   Error initializing runtime service opteed_fast
INFO:    BL3-1: Preparing for EL3 exit to normal world		INFO:    BL3-1: Preparing for EL3 exit to normal world
INFO:    BL3-1: Next image address = 0x1000000			INFO:    BL3-1: Next image address = 0x1000000
INFO:    BL3-1: Next image spsr = 0x3c9				INFO:    BL3-1: Next image spsr = 0x3c9
 								 
 								 
U-Boot 2015.01-g2bd1e3b (Apr 09 2022 - 06:14:31), Build: jenk	U-Boot 2015.01-g2bd1e3b (Apr 09 2022 - 06:14:31), Build: jenk
 								 
DRAM:  2 GiB							DRAM:  2 GiB
Relocation Offset is: 76edf000					Relocation Offset is: 76edf000
gpio: pin GPIODV_25 (gpio 44) value is 1			gpio: pin GPIODV_25 (gpio 44) value is 1
register usb cfg[0][1] = 0000000077f5d610			register usb cfg[0][1] = 0000000077f5d610
[CANVAS]canvas init						[CANVAS]canvas init
vpu: error: vpu: check dts: FDT_ERR_BADMAGIC, load default pa	vpu: error: vpu: check dts: FDT_ERR_BADMAGIC, load default pa
[... 216 lines cut ...]
							      >	store scrub off|partition size
							      >		scrub the area from offset and size 
							      >	store dtb iread/read/write addr <size>
							      >		read/write dtb, size is optional 
							      >	store key read/write addr <size>
							      >		read/write key, size is optional 
							      >	store mbr addr
							      >	   update mbr/partition table by dtb
							      >	 
							      >	cmd store failed 
							      >	Err imgread(L135):Fail to read 0x100000B from part[recovery] 
							      >	gxl_p211_v1#
```

</p>
</details>

```
mmc env offset: 0x7400000 					mmc env offset: 0x7400000 
							      >	*** Warning - bad CRC, using default environment
							      >	 
							      >	Saving Environment to aml-storage...
							      >	mmc env offset: 0x7400000 
							      >	Writing to MMC(1)... done
```

ophub, 2023-05-02:
openwrt在系统，晶晨宝盒，插件设置，勾选上写入bootloader ，再写入即可。

Becods, 2023-05-02:
> openwrt在系统，晶晨宝盒，插件设置，勾选上写入bootloader ，再到安装openwrt里选择写入即可。

已勾选并写入，但仍无法启动

<details><summary>写入日志</summary>
<p>

```
root@OpenWrt:/# cat /tmp/amlogic/amlogic_check_install.log 
AUTO_MAINLINE_UBOOT: yes
SPECIFY_DTB: 
SPECIFY_SOC: 
SPECIFY_UBOOT: 
SHARED_FSTYPE: btrfs
Current device: Amlogic Meson GXL (S905L3b) MBH-M302A Box [ amlogic ]
/dev/mmcblk2
mmcblk2 : 15634268160 bytes
ROOTFS: sda2
BOOT: sda1
Please select s9xxx box model:
----------------------------------------------------------------------------------------------------
 101. Phicomm-N1                    s905d   2GB-Mem,1Gb-Nic,brcm43455-wifi
 102. Phicomm-N1(DMA-thresh)        s905d   When-ethmac-flow-control-is-off
 103. HG680P                        s905x   2+8G,100Mb-Nic
 104. TX3-Mini                      s905w   2+8G,100Mb-Nic
 105. MECOOL-KI-Pro                 s905d   2G/16G,1Gb-Nic
 106. T95                           s905x   2G/16G,1Gb-Nic
 107. B860H                         s905x   2+8G,100Mb-Nic
 108. TBee                          s905x   2+8G,100Mb-Nic
 109. W95                           s905w   2GB-RAM,16GB-ROM,100Mb-Nic
 110. X96-Mini                      s905w   2+8G,100Mb-Nic
 111. TX9                           s905x   100Mb-Nic,rtl8723cs-WLAN
 112. M302A,M304A,CM211-1,CM311-1   s905l3b 2+8G,100Mb-Nic
 113. MGV2000,MGV3000,M301A         s905l2  2+8G,100Mb-Nic
 114. Wojia-TV-IPBS9505             s905l2  2+8G,100Mb-Nic
 115. Q96-mini                      s905lb  1G/8G,100Mb-Nic
 116. CM311-1,HG680-LC,M401A        s905l3  2+8G,100Mb-Nic
 117. E900V22E                      s905l3b 2+8G,100Mb-Nic
 118. BesTV-R3300L,SumaVision-Q7    s905lb  2+8G,100Mb-Nic
 119. X96W,FunTV,MXQ-Pro-4K         s905w   2GB-RAM,16GB-ROM,100Mb-Nic,AP6255-wifi/BT
 120. E900V22D                      s905l3b 2+8G,100Mb-Nic
 121. UNT400G1                      s905l3  2+8G,100Mb-Nic
 122. E900v21E                      s905l2  2+8G,100Mb-Nic
 123. IP103H,TY1608                 s905l3b 2+8G,100Mb-Nic
 124. CM211-1                       s905l3  2+8G,100Mb-Nic
 201. Octopus-Planet                s912    2GB-Mem,1Gb-Nic
 202. Octopus-Planet(FAKE)          s912    2GB-Mem,1Gb-Nic
 203. H96-Pro-Plus                  s912    2G/32G,1Gb-Nic
 204. Tanix-TX92                    s912    3GB-DDR4,32GB-eMMC,5G-WIFI,1Gb-Nic
 205. VORKE-Z6-Plus                 s912    3GB-DDR3,32GB-eMMC,1Gb-Nic
 206. MECOOL-M8S-Pro-L              s912    2G/3G-RAM,16G/32G-ROM,100Mb-Nic
 207. T95Z-Plus                     s912    3G+32G,1Gb-Nic
 208. TX9-Pro(3G_32G_1Gb)           s912    3G-RAM,32G-ROM,1Gb-Nic,brcm43455-wifi
 209. TX9-Pro(2G_16G_100Mb)         s912    3G-RAM,32G-ROM,100Mb-Nic,RTL8189ETV-wifi
 210. Nexbox-A1,A95X                s912    2G-DDR3-RAM,16G-eMMC,1Gb-Nic
 211. Nexbox-A95X-A2                s912    2GB-RAM,16GB-ROM,2.4G/5.0G-WiFi,1Gb-Nic
 212. Tanix-TX8-MAX                 s912    3GB-RAM,16GB/32GB-eMMC,1Gb-Nic
 213. Vontar-X92                    s912    3GB-RAM,16GB/32GB-eMMC,1Gb-Nic
 214. Phicomm-T1                    s912    2G-RAM,16G-ROM,100Mb-Nic,Wifi,Bluetooth
 301. X96-Max-4GB,Tx5-Max           s905x2  4GB-Mem,1Gb-Nic
 302. X96-Max-2GB,A95X-F2           s905x2  2GB-Mem,100Mb-Nic
 303. MECOOL-KM3-4G                 s905x2  4+64G/128G,100Mb-Nic
 304. E900V22C-D                    s905l3a 4+64G/128G,100Mb-Nic
 305. CM311-1a-YST                  s905l3a 2+16G,BT-rtl8761b,100Mb-Nic
 306. M401A,UNT403A,B863AV3.2-M     s905l3a 2+16G,100Mb-Nic
 401. GT-King                       s922x   4GB-Mem,1Gb-Nic,brcm4356-wifi
 402. GT-King-Pro,X88-King          s922x   4GB-Mem,1Gb-Nic,brcm4356-wifi
 403. GT-King-Pro-H                 s922x   4GB-Mem,1Gb-Nic,brcm4356-wifi
 404. GT-King-Pro-Rev_A             s922x   4GB-Mem,1Gb-Nic,brcm4356-wifi
 405. Hardkernel-ODroid-N2          s922x   4GB-Mem,1Gb-Nic
[... 477 lines cut ...]
							      >	store dtb iread/read/write addr <size>
							      >		read/write dtb, size is optional 
							      >	store key read/write addr <size>
							      >		read/write key, size is optional 
							      >	store mbr addr
							      >	   update mbr/partition table by dtb
							      >
							      >	cmd store failed 
							      >	Err imgread(L135):Fail to read 0x100000B from part[recovery] 
							      >	gxl_p211_v1#
```

</p>
</details>

依旧是crc错误

```
mmc env offset: 0x7400000 					mmc env offset: 0x7400000 
							      >	*** Warning - bad CRC, using default environment
							      >
							      >	Saving Environment to aml-storage...
							      >	mmc env offset: 0x7400000 
							      >	Writing to MMC(1)... done
```

ophub, 2023-05-02:
复制一下u-boot

Becods, 2023-05-02:
> U盘中复制一下u-boot。先复制好后，再写入
> 
> ```
> cd /boot
> copy -f u-boot-s905x-s912.bin u-boot.emmc
> copy -f u-boot-s905x-s912.bin u-boot.ext
> ```

依旧不行

<details><summary>uboot日志</summary>
<p>

```
GXLX2:BL1:3cfee7:42a5ae;FEAT:ADFD318C:0;POC:3;RCY:0;EMMC:0;READ:0;0.0;CHK:0;
TE: 162615

BL2 Built : 13:35:42, Jun  1 2021. 
gxl g49d5acc - liqiang.hou@droid07-sz

Board ID = 3, adc=225
set vcck to 1070 mv
set vddee to 1070 mv
CPU clk: 1200MHz

DDR4 chl: Rank0+1 @ 912MHz - FAIL
DDR4 chl: Rank0 @ 912MHz
bist_test rank: 0 1e 08 35 32 1a 4b 18 00 31 34 1a 4f 13 00 26 38 20 51 16 00 2c 38 20 51 616   - PASS

Rank0: 2048MB(auto)-2T-18
AddrBus test pass!
-s
Load fip header from eMMC, src: 0x0000c200, des: 0x01400000, size: 0x00004000
New fip structure!
Load bl30 from eMMC, src: 0x00010200, des: 0x013c0000, size: 0x00007600
Load bl301 from eMMC, src: 0x00018200, des: 0x01380000, size: 0x00002600
Load bl31 from eMMC, src: 0x0001c200, des: 0x10100000, size: 0x00019600
Load bl33 from eMMC, src: 0x00038200, des: 0x01000000, size: 0x0004da00
NOTICE:  BL3-1: v1.0(debug):27617c8a4
NOTICE:  BL3-1: Built : 18:02:23, Aug 18 2021
NOTICE:  BL31: GXL normal boot!
NOTICE:  BL31: BL33 decompress pass
[Image: gxl_v1.1.3103-923400455-dirty 2020-12-28 18:42:52 xingyu.chen@droid12-sz]
efuse init ops = c5
efuse init hdcp = c, cf9=7
x2_hp_e = 0
bl30: check_permit, count is 1
bl30: check_permit: ok!
chipid: 0 0 3 c c 0 f1 b3 40 a0 0 c5 not ES chip
[0.703069 Inits done]
INFO:    BL3-1: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL3-1: Preparing for EL3 exit to normal world
INFO:    BL3-1: Next image address = 0x1000000
INFO:    BL3-1: Next image spsr = 0x3c9

U-Boot 2015.01-g2bd1e3b (Apr 09 2022 - 06:14:31), Build: jenkins-FVOS-git_FVOS_amlogic_9_CMCCHUBEI_PRIV-190

DRAM:  2 GiB
Relocation Offset is: 76edf000
gpio: pin GPIODV_25 (gpio 44) value is 1
register usb cfg[0][1] = 0000000077f5d610
[CANVAS]canvas init
vpu: error: vpu: check dts: FDT_ERR_BADMAGIC, load default parameters
vpu: clk_level = 7
vpu: set clk: 666667000Hz, readback: 666660000Hz(0x300)
vpp: vpp_init
MMC:   aml_priv->desc_buf = 0x0000000073ecfac0
aml_priv->desc_buf = 0x0000000073ed1de0
[... 532 lines cut ...]
Successfully mounted
Create folder ... 
Create subvolume './etc'
complete.
Copy data ... 
copy [ root ] ...
copy [ etc ] ...
copy [ bin ] ...
copy [ sbin ] ...
copy [ lib ] ...
copy [ opt ] ...
copy [ usr ] ...
copy [ www ] ...
Copy complete.
Edit configuration file ...
../../lib/modules/5.15.109-ophub/kernel/drivers/net/ethernet/qualcomm/sfe/shortcut-fe/shortcut-fe-cm.ko
../../lib/modules/5.15.109-ophub/shortcut-fe-cm.ko
Create initial etc snapshot -> .snapshots/etc-000Create a readonly snapshot of 'etc' in '.snapshots/etc-000'
complete.
Create a shared file system.
Successful installed, please unplug the USB, re-insert the power supply to start the openwrt.
```

</p>
</details>

ophub, 2023-05-02:
chmod加一下执行权限，看上面贴子里更新的命令。

Becods, 2023-05-02:
> chmod加一下执行权限，看上面贴子里更新的命令。

依旧不行

uboot日志在复制前后并无变化

<details><summary>uboot日志</summary>
<p>

```
GXLX2:BL1:3cfee7:42a5ae;FEAT:ADFD318C:0;POC:3;RCY:0;EMMC:0;READ:0;0.0;CHK:0;
TE: 157454

BL2 Built : 13:35:42, Jun  1 2021. 
gxl g49d5acc - liqiang.hou@droid07-sz

Board ID = 3, adc=225
set vcck to 1070 mv
set vddee to 1070 mv
CPU clk: 1200MHz

DDR4 chl: Rank0+1 @ 912MHz - FAIL
DDR4 chl: Rank0 @ 912MHz
bist_test rank: 0 1f 08 36 32 1a 4b 18 00 30 35 1a 50 13 00 26 39 21 51 16 00 2c 38 20 51 616   - PASS

Rank0: 2048MB(auto)-2T-18
AddrBus test pass!
-s
Load fip header from eMMC, src: 0x0000c200, des: 0x01400000, size: 0x00004000
New fip structure!
Load bl30 from eMMC, src: 0x00010200, des: 0x013c0000, size: 0x00007600
Load bl301 from eMMC, src: 0x00018200, des: 0x01380000, size: 0x00002600
Load bl31 from eMMC, src: 0x0001c200, des: 0x10100000, size: 0x00019600
Load bl33 from eMMC, src: 0x00038200, des: 0x01000000, size: 0x0004da00
NOTICE:  BL3-1: v1.0(debug):27617c8a4
NOTICE:  BL3-1: Built : 18:02:23, Aug 18 2021
NOTICE:  BL31: GXL normal boot!
NOTICE:  BL31: BL33 decompress pass
[Image: gxl_v1.1.3103-923400455-dirty 2020-12-28 18:42:52 xingyu.chen@droid12-sz]
efuse init ops = c5
efuse init hdcp = c, cf9=7
x2_hp_e = 0
bl30: check_permit, count is 1
bl30: check_permit: ok!
chipid: 0 0 3 c c 0 f1 b3 40 a0 0 c5 not ES chip
[0.698781 Inits done]
INFO:    BL3-1: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL3-1: Preparing for EL3 exit to normal world
INFO:    BL3-1: Next image address = 0x1000000
INFO:    BL3-1: Next image spsr = 0x3c9

U-Boot 2015.01-g2bd1e3b (Apr 09 2022 - 06:14:31), Build: jenkins-FVOS-git_FVOS_amlogic_9_CMCCHUBEI_PRIV-190

DRAM:  2 GiB
Relocation Offset is: 76edf000
gpio: pin GPIODV_25 (gpio 44) value is 1
register usb cfg[0][1] = 0000000077f5d610
[CANVAS]canvas init
vpu: error: vpu: check dts: FDT_ERR_BADMAGIC, load default parameters
vpu: clk_level = 7
vpu: set clk: 666667000Hz, readback: 666660000Hz(0x300)
vpp: vpp_init
MMC:   aml_priv->desc_buf = 0x0000000073ecfac0
aml_priv->desc_buf = 0x0000000073ed1de0
SDIO Port B: 0, SDIO Port C: 1
emmc/sd response timeout, cmd8, status=0x1ff2800
emmc/sd response timeout, cmd55, status=0x1ff2800
init_part() 278: PART_TYPE_DOS
[... 549 lines cut ...]
complete.
Wait for the rootfs file system to mount ... 
Successfully mounted
Create folder ... 
Create subvolume './etc'
complete.
Copy data ... 
copy [ root ] ...
copy [ etc ] ...
copy [ bin ] ...
copy [ sbin ] ...
copy [ lib ] ...
copy [ opt ] ...
copy [ usr ] ...
copy [ www ] ...
Copy complete.
Edit configuration file ...
Create initial etc snapshot -> .snapshots/etc-000Create a readonly snapshot of 'etc' in '.snapshots/etc-000'
complete.
Create a shared file system.
Successful installed, please unplug the USB, re-insert the power supply to start the openwrt.
```

</p>
</details>

ophub, 2023-05-02:
换6.1内核的试试

Becods, 2023-05-02:
> 换6.1内核的试试

无法启动，不论是openwrt还是armbian

```

scanning bus 0 for devices... 2 USB Device(s) found
       scanning usb for storage devices... init_part() 278: PART_TYPE_DOS
1 Storage Device(s) found
reading s905_autoscript
1654 bytes read in 53 ms (30.3 KiB/s)
## Executing script at 01020000
start amlogic old u-boot
## Error: "bootfromsd" not defined
** Bad device mmc 0 **
reading boot_android
** Unable to read file boot_android **
** Bad device mmc 0 **
reading u-boot.ext
650183 bytes read in 474 ms (1.3 MiB/s)
## Starting application at 0x01000000 ...
[BL31]: tee size: 0
[BL31]: tee size: 0
INFO:    PSCI Affinity Map:
INFO:      AffInst: Level 0, MPID 0x0, State ON
INFO:      AffInst: Level 0, MPID 0x1, State OFF
INFO:      AffInst: Level 0, MPID 0x2, State OFF
INFO:      AffInst: Level 0, MPID 0x3, State OFF
bl31 reboot reason: 0x1
bl31 reboot reason: 0x0
ERROR:   gxbb System Reset: SCP error 16.
```

<details><summary>启动日志</summary>
<p>

```
GXLX2:BL1:3cfee7:42a5ae;FEAT:ADFD318C:0;POC:3;RCY:0;EMMC:0;READ:0;0.0;CHK:0;
TE: 100376

BL2 Built : 13:35:42, Jun  1 2021. 
gxl g49d5acc - liqiang.hou@droid07-sz

Board ID = 3, adc=225
set vcck to 1070 mv
set vddee to 1070 mv
CPU clk: 1200MHz

DDR4 chl: Rank0+1 @ 912MHz - FAIL
DDR4 chl: Rank0 @ 912MHz
bist_test rank: 0 1f 08 36 34 1b 4d 18 00 31 35 1a 50 13 00 27 3a 22 52 16 00 2d 38 20 51 616   - PASS

Rank0: 2048MB(auto)-2T-18
AddrBus test pass!
-s
Load fip header from eMMC, src: 0x0000c200, des: 0x01400000, size: 0x00004000
New fip structure!
Load bl30 from eMMC, src: 0x00010200, des: 0x013c0000, size: 0x00007600
Load bl301 from eMMC, src: 0x00018200, des: 0x01380000, size: 0x00002600
Load bl31 from eMMC, src: 0x0001c200, des: 0x10100000, size: 0x00019600
Load bl33 from eMMC, src: 0x00038200, des: 0x01000000, size: 0x0004da00
NOTICE:  BL3-1: v1.0(debug):27617c8a4
NOTICE:  BL3-1: Built : 18:02:23, Aug 18 2021
NOTICE:  BL31: GXL normal boot!
NOTICE:  BL31: BL33 decompress pass
[Image: gxl_v1.1.3103-923400455-dirty 2020-12-28 18:42:52 xingyu.chen@droid12-sz]
efuse init ops = c5
efuse init hdcp = c, cf9=7
x2_hp_e = 0
bl30: check_permit, count is 1
[... 163 lines cut ...]
1654 bytes read in 53 ms (30.3 KiB/s)
## Executing script at 01020000
start amlogic old u-boot
## Error: "bootfromsd" not defined
** Bad device mmc 0 **
reading boot_android
** Unable to read file boot_android **
** Bad device mmc 0 **
reading u-boot.ext
650183 bytes read in 474 ms (1.3 MiB/s)
## Starting application at 0x01000000 ...
[BL31]: tee size: 0
[BL31]: tee size: 0
INFO:    PSCI Affinity Map:
INFO:      AffInst: Level 0, MPID 0x0, State ON
INFO:      AffInst: Level 0, MPID 0x1, State OFF
INFO:      AffInst: Level 0, MPID 0x2, State OFF
INFO:      AffInst: Level 0, MPID 0x3, State OFF
bl31 reboot reason: 0x1
bl31 reboot reason: 0x0
ERROR:   gxbb System Reset: SCP error 16.
```

</p>
</details>

ophub, 2023-05-02:
看来这个uboot不适合你，很多盒子都有同名不同芯的差异。自己试吧，就那几个，总有一个适合你。

Becods, 2023-05-02:
在下载了接近30g的安卓固件，刷入20+固件之后，总算找到个能成功将openwrt刷入emmc并启动中的了

[image]

此贴终了

closed 2023-05-02 by Becods

ophub, 2023-05-02:
分享一下你用的安卓系统的下载，我备份一个

Becods, 2023-05-02:
> 分享一下你用的安卓系统的下载，我备份一个

上传完成发现阿里云盘扫文件头而不是文件后缀

在重新上传自解压

待会更新

ophub, 2023-05-02:
这个exe是自解压文件？还是改了后缀名了？

Becods, 2023-05-02:
> 这个exe是自解压文件？还是改了后缀名了？

自解压，右键直接用7z打开即可，或者双击打开解压

阿里网盘依据文件头而不是文件后缀判断是否能分享，只能打包成自解压文件

ophub, 2023-05-02:
好的，我保存到我的网盘了，节后我转存到： https://github.com/ophub/kernel/releases/tag/tools