ophub/amlogic-s9xxx-openwrt #425: O大，编译的固件不显示频率温度和接口信息

boyhaibo, 2023-05-03:
O大你好，我编译的固件打包出来，启动不显示频率温度和接口信息，是不是有什么依赖没选
如果要显示这些信息需要添加那些依赖，O大能告知一下吗，非常感谢！
我添加了以下依赖
CONFIG_PACKAGE_perl=y
CONFIG_PACKAGE_perl-http-date=y
CONFIG_PACKAGE_perlbase-file=y
CONFIG_PACKAGE_perlbase-getopt=y
CONFIG_PACKAGE_perlbase-time=y
CONFIG_PACKAGE_perlbase-unicode=y
CONFIG_PACKAGE_perlbase-utf8=y
CONFIG_PACKAGE_blkid=y
CONFIG_PACKAGE_fdisk=y
CONFIG_PACKAGE_lsblk=y
CONFIG_PACKAGE_parted=y
CONFIG_PACKAGE_attr=y
CONFIG_PACKAGE_btrfs-progs=y
CONFIG_BTRFS_PROGS_ZSTD=y
CONFIG_PACKAGE_chattr=y
CONFIG_PACKAGE_dosfstools=y
CONFIG_PACKAGE_e2fsprogs=y
CONFIG_PACKAGE_f2fs-tools=y
CONFIG_PACKAGE_f2fsck=y
CONFIG_PACKAGE_lsattr=y
CONFIG_PACKAGE_mkf2fs=y
CONFIG_PACKAGE_xfs-fsck=y
CONFIG_PACKAGE_xfs-mkfs=y
CONFIG_PACKAGE_bsdtar=y
CONFIG_PACKAGE_pigz=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_gawk=y
CONFIG_PACKAGE_getopt=y
CONFIG_PACKAGE_losetup=y
CONFIG_PACKAGE_tar=y
CONFIG_PACKAGE_uuidgen=y
CONFIG_PACKAGE_acpid=y
CONFIG_PACKAGE_kmod-brcmfmac=y
CONFIG_BRCMFMAC_SDIO=y
CONFIG_PACKAGE_kmod-brcmutil=y
CONFIG_PACKAGE_kmod-cfg80211=y
CONFIG_PACKAGE_kmod-mac80211=y
CONFIG_PACKAGE_hostapd-common=y
CONFIG_PACKAGE_wpa-cli=y
CONFIG_PACKAGE_wpad-basic=y
CONFIG_PACKAGE_iw=y
CONFIG_PACKAGE_ntfs3-mount=y
CONFIG_PACKAGE_coreutils=y
CONFIG_PACKAGE_coreutils-base64=y
CONFIG_PACKAGE_coreutils-nohup=y
CONFIG_PACKAGE_jq=y
CONFIG_PACKAGE_pv=y
[image]

boyhaibo, 2023-05-03:
我上面列出的依赖就是10.11里面的

ophub, 2023-05-03:
ttyd里面看看显示情况

boyhaibo, 2023-05-03:
ttyd里面的显示正常的，有频率和温度信息，就是luci界面不显示

ophub, 2023-05-03:
换别的皮肤

boyhaibo, 2023-05-03:
还是不显示，算了不弄了不影响使用