ophub/amlogic-s9xxx-openwrt #426: mgv2000更新内核失败

JayceeNee, 2023-05-05:
**Device Information | 设备信息**
 - SOC: s905l2
 - Model mgv2000

**Armbian Version | 系统版本**
 - Kernel Version: 5.15.109-ophub
 - Release: OpenWrt SNAPSHOT r22711-4fa9ec125a / LuCI Master git-23.104.28202-588c6a1

**Describe the bug | 问题描述**
在线更新内核失败，提示[%1B[1;91m Error %1B[0m] Missing [ tar ] in OpenWrt firmware, unable to update kernel，但系统已有tar

[image]
[image]

JayceeNee, 2023-05-05:
尝试完整更新全系统，提示 The device name is empty and cannot be recognized.手动上传更新情况相同
[image]

ophub, 2023-05-05:
你的界面不正常，怎么重复显示2次内核版本呢？这几个按钮不要一起点，只执行一个。

现在切换到手动上传更新那个里，把看到的这些tar.gz文件都删了，重启一下，重新在线下载更新

JayceeNee, 2023-05-05:
删除并重启后 在线更新情况依旧
[image]

JayceeNee, 2023-05-06:
返回如图
[image]

ophub, 2023-05-06:
正常情况下他应该这样显示，你的显示双重，不知道是不是你的luci皮肤导致的，你在系统里切换个别的皮肤看看。

[image]

closed 2023-05-07 by ophub