ophub/amlogic-s9xxx-openwrt #428: Help Installing on non branded s905x3, 2g, 100m device

cloudomate, 2023-05-07:
hi I have a amlogic based device with following specs: s905x3, 2G, 100m ethernet and soldered marvel usb wifi 88w8897, it was gifted at an event to i believe its custom built.  I have UART console access to this device and therefore get to bootloader. 
1. sdcard/usb method does not work as there is no reset button which can force device to boot from usb/sd card.  
2. Still i have tried just inserting the sdcard (after flashing it with release version for s905x3). I extracted my box dtb file and copied to boot partition /dtb/amlogic folder and modified the uenv.txt with file name. here is the log 
Name: SDTran Speed: 50000000
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: (0x3a4400000 Bytes) 14.6 GiB
mmc clock: 40000000
Bus Width: 4-bit
emmc/sd read error, cmd17, cmd->cmdarg=0x0, status=0x1ff200d
retry read, count: 5
emmc/sd read error, cmd17, cmd->cmdarg=0x0, status=0x1ff200f
retry read, count: 4
emmc/sd read error, cmd17, cmd->cmdarg=0x0, status=0x1ff200f
retry read, count: 3
emmc/sd read error, cmd17, cmd->cmdarg=0x0, status=0x1ff200d
retry read, count: 2
emmc/sd read error, cmd17, cmd->cmdarg=0x0, status=0x1ff200d
retry read, count: 1
emmc/sd read error, cmd17, cmd->cmdarg=0x0, status=0x1ff200d
retry read error !!!
** Unrecognized filesystem type **
Its formatted correctly so i believe 
3. Since i have access to the bootloader and can make use of tftp to copy files over or even modify the bootloader env variables , is there any guide which can help ?

ophub, 2023-05-07:
Please refer to the [Armbian Documentation](https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/documents) for more information on Armbian.

closed 2023-05-07 by ophub

cloudomate, 2023-05-11:
Thanks i realized the issue was because of sdcard.