ophub/amlogic-s9xxx-openwrt #429: Firewall4支持

0xDarker, 2023-05-09:
请问作者大佬，预编译固件能默认开启Firewall4吗？
或者自己编译需要修改哪些选项。

ophub, 2023-05-09:
OpenWrt 22.03 将 Firewall4 作为新的基于 Nftables 的默认防火墙，并替代了他们旧的基于 IPTables 的“Firewall3”路由器。Nftables 支持的 Firewall4 应该可以作为替代品，并很好地转换现有的防火墙规则。方法开官方文档。

https://openwrt.org/docs/guide-user/firewall/firewall_configuration

0xDarker, 2023-05-09:
> 

不是firewall4的使用，我看config文件里面[# CONFIG_PACKAGE_firewall4 is not set](https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/config-openwrt/lede-master/config)。然后安装基于firewall4的ipk时报错：
```bash
bash-5.2# opkg install luci-app-xray-fw4_2.0.0-1_all.ipk
Unknown package 'luci-app-xray-fw4'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency firewall4 for luci-app-xray-fw4
 * pkg_hash_check_unresolved: cannot find dependency kmod-nft-tproxy for luci-app-xray-fw4
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-xray-fw4 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-xray-fw4.

```
想问下这是编译固件的时候是没开启firewall4吗，如果自己编译需要修改哪些config项?
我是armbian kernel 6.1上docker装的openwrt

0xDarker, 2023-05-09:
> 深居庭院，莫议萧墙。 不要发看不懂的插件名字。插件的集成和使用方法都相同。

OKOK，不管插件吧，就是那个报错 cannot find dependency firewall4 和 cannot find dependency kmod-nft-tproxy 这个怎么弄呢，这应该是报的内核里没这模块吧

ophub, 2023-05-09:
kmod后期无法安装，只能在编译时选择上集成进本。

0xDarker, 2023-05-09:
> kmod后期无法安装，只能在编译时选择上集成进本。config里加上他

感谢，那我只开CONFIG_PACKAGE_firewall4=y和CONFIG_PACKAGE_kmod-nft-tproxy=y就够了吗，不太了解还有没有别的依赖需要打开的所以来问问。

ophub, 2023-05-09:
就你安装ipk时错误提示缺少的添加上即可。

closed 2023-05-18 by ophub