ophub/amlogic-s9xxx-openwrt #434: ct2000 指定内核无效

OwO-GitHub, 2023-05-17:
**Device Information | 设备信息**
 - SOC: S922X
 - Model CT2000

**OpenWrt Version | 系统版本**
 - Kernel Version: ALL
 - Release: immortalwrt

**Describe the bug | 问题描述**
Action编译的时候不管是指定flippy 还是stable，最后下载内核的时候，总是会下载dev内核。
[image]
[image]

ophub, 2023-05-17:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/make-openwrt/openwrt-files/common-files/etc/model_database.conf

在KERNEL_TAGS里指定为 stable, 5.x.y, 6.x.y 时可以被自选 tags 替换。

指定为 rk3588, flippy, dev 时不可被替换。

OwO-GitHub, 2023-05-18:
好的，知道了谢谢。

closed 2023-05-18 by OwO-GitHub