ophub/amlogic-s9xxx-openwrt #435: e900v21e 机顶盒刷openwrt  电脑无法与机顶盒LAN口连接

xss-1986, 2023-05-23:
O大，用你的固件刷了s905l2的芯片，系统装载到u盘了，但是用网线连接机顶盒，电脑显示物理网口拔出。是什么问题，另外能通过命令方式开启机顶盒的WiFi吗，小白一个

ophub, 2023-05-23:
看一下文档的说明，把电脑手动设置到192.168.1.x同网段

xss-1986, 2023-05-23:
因为电脑是w10笔记本，尝试过关掉无线，对有线网络设置固定IP同网段，但是在ping  机顶盒的时候电脑还是显示无线网段，数据发送，能接受，就是无法访问

xss-1986, 2023-05-23:
> 因为电脑是w10笔记本，尝试过关掉无线，对有线网络设置固定IP同网段，但是在ping 机顶盒的时候电脑还是显示无线网段，数据发送，能接受，就是无法访问

> 因为电脑是w10笔记本，尝试过关掉无线，对有线网络设置固定IP同网段，但是在ping 机顶盒的时候电脑还是显示无线网段，数据发送，能接受，就是无法访问

xss-1986, 2023-05-23:
我看之前有人向您提问说，因为刷的机顶盒固件的缘故，可能会导致机顶盒和电脑连接，始终无法访问op默认ip: 192.168.1.1的网址，或许我这个也是这情况？后面我电脑接口、机顶盒接口、网线都测试都没问题，按理电脑用网线连接到机顶盒上，应该有所反应

xss-1986, 2023-05-23:
当然我ifconfig命令去调取出现的是一个br-lan\ docker0\ eth0\ lo  这四个link encap

ophub, 2023-05-23:
同网段，网线直联的，流览器不能访问，这不合理。

电脑只填2处即可，
ip: 192.168.1.2
子网：255.255.255.0
其他空白

xss-1986, 2023-05-23:
然后br-lan 里有我设置更改的inet addr:192.168.1.3 ,这个我看其他人说不要跟路由器冲突，然后更改的
主要是电脑和机顶盒连接没有反馈信息，不知道咋整

xss-1986, 2023-05-23:
非科学上网，反馈慢，我自己只能在琢磨琢磨，物理有线我手动设置网段，但是在ping 的时候反馈信息还是无线的，哎小白碰壁。不过在这个过程我用老电脑xp也去设置过，同网段，网卡显示物理拔出

ophub, 2023-05-23:
把电脑的无线自动联接勾取消了，设置忽略此wifi(你的无线id)

xss-1986, 2023-05-23:
我想说网络问题，我发现了，U盘未插入，电脑和机顶盒能连接，插入U盘启动机顶盒后，电脑和机顶盒就没办法连接，电脑显示物理网线接口拔出。难道是U盘的锅？

ophub, 2023-05-23:
把盒子插在路由器，启动安卓系统，adb shell先启用了U盘更新