ophub/amlogic-s9xxx-openwrt #437: No wired network X96 Mini K5.15.113

Tasmoto, 2023-05-27:
Device Information

SOC: [S905w]
Model: [X96 mini]
OpenWrt Version

Kernel Version: 5.15.113-ophub
Release: official

Describe the bug

I follow the steps to build the images with all the default data (sudo ./make -b s905w_s905w-x96-mini)
I use with Rufus the file "openwrt_amlogic_s905w-x96-mini_k5.15.113_2023.05.27.img.gz"
I boot from the SD card and it boots fine, I have console, but no network interface.

I look for information:

ubus list network.interface.*	->	network.interface.lan
						network.interface.loopback

ifstatus lan			->	up: false
						pending: false
						avaliable: false
						autostart: true
						dynamic: false
						proto: static
						device: br-lan
						data:

						errors: subsystem: interface
							   code: NO_DEVICE

What could I do to get it to up the wired network card?

I am trying to build with kernel 5.15.113 because with the older version (downloaded from github) I have dependency problems with various packages like WireGuard.

It also surprises me that the image compiled on github usually weighs 277Mb and the image I compile weighs less, 161Mb.

ophub, 2023-05-27:
use openwrt.tar.gz

Tasmoto, 2023-05-27:
I use the file "openwrt-armvirt-64-default-rootfs.tar.gz" downloaded from "https://openwrt.tetaneutral.net/snapshots/targets/armvirt/64/"

Do I need that file from another location or specific version?

ophub, 2023-05-27:
Can use

closed 2023-05-30 by ophub