ophub/amlogic-s9xxx-openwrt #441: 本地化打包时出错

13528943945, 2023-06-10:
**Device Information | 设备信息**
 - SOC: [e.g. S095x3]
 - Model [e.g. TX3]

**OpenWrt Version | 系统版本**
 - Kernel Version: [e.g. 5.15.100]
 - Release: [e.g. immortalwrt]

**Describe the bug | 问题描述**  
本地化打包时出错,日志如下：
[ STEPS ] Start making OpenWrt firmware...
(1.1) Start making OpenWrt [ s905x3 - stable/6.1.33 ]. Remaining space is 4G.
 [🌿] (1/6) Make OpenWrt image.
 [🌿] (2/6) Extract OpenWrt files.
mount: /home/lyh/amlogic-s9xxx-openwrt/tmp/6.1.33/s905x3/rootfs: wrong fs type, bad option, bad superblock on /dev/loop5p2, missing codepage or helper program, or other error.

是否我自己编译的ROOTFS缺少某个组件，比如 codepage

ophub, 2023-06-10:
首页，本地依赖包，先安装了再打包。

重新打包前先把根目录下创建的tmp目录删了

13528943945, 2023-06-10:
果然是本地UBUNTU虚拟机缺少相关依赖，安装后，打包成功了。多谢大佬！

ophub, 2023-06-10:
好的

closed 2023-06-13 by ophub