ophub/amlogic-s9xxx-openwrt #442: S905X-T95 cannot access Web Gui Luci ?

Biangkerok32, 2023-06-11:
**Device Information | 设备信息**
 - SOC: [S095x-t95]

**OpenWrt Version | 系统版本**
 - Kernel Version: [last]
 - Release: [lede/immortalwrt]

**Describe the bug | 问题描述**
I'm try use last build lede and immortal with last kernel,but I'm can't access luci web GUI?
I can access ssh work it.

I try enter "ps -w" on ssh terminal,but i see "usr/sbin/uhttpd -f -h /www -r HOSTNAME -x /cgi-bin -t 60 -T 30 -k 20 -A 1 -n 3 -N 100 -R" without "0.0.0.0:80"

How to fix that 🤔

Thanks.

ophub, 2023-06-11:
Screenshot of the complete interface of the terminal

ophub, 2023-06-12:
```
rm -rf /tmp/luci-*
sync
reboot
```

Biangkerok32, 2023-06-12:
It's fix now . 👍

I'm open Luci Web GUI with other browser 

I thinks error from my browser cache.

[image]

closed 2023-06-12 by Biangkerok32