ophub/amlogic-s9xxx-openwrt #444: 中国移动新魔百盒M201-S用哪个刷机包？刷过的盆友请指教一下，谢谢！

hzlyu123, 2023-06-16:
中国移动新魔百盒M201-S用哪个刷机包？刷过的盆友请指教一下，谢谢！

wordlesswind, 2023-06-27:
看了看，SoC 为 `Amlogic S905L`，目前没有明确支持 M201-S，没有 `dtb` 文件。

你可以尝试刷入 [openwrt_lede_amlogic_s905l_k6.1.34_2023.06.22.img.gz](https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_lede_save_2023.06/openwrt_lede_amlogic_s905l_k6.1.34_2023.06.22.img.gz)。

---

我找到了一些来自[人中日月](https://space.bilibili.com/97745521/)的用于 EmuELEC 的 dtb 文件：[device_trees](https://pan.emuelec.cn/Emuelec_cn/integrate/Emuelec%20S905L/device_trees)

此外我还从 [EMUELEC-4G.img](https://pan.emuelec.cn/Emuelec_cn/integrate/Emuelec%20S905L/EMUELEC%203.3%204G/EMUELEC-4G.img) 提取了 Boot 分区文件，[据博主人中日月称](https://www.bilibili.com/read/cv4420521/)：“_S905/S912这次镜像是在S905L盒子上制作的，所以2018年上半年以前的S905L/M2的机器（包括但不限于：数码视讯Q7、部分数码视讯Q5、魔百盒HM201、百视通R3300L、咪咕MG101、魔百盒M201S/M201D、部分创维E900V21C）不用替换dtb直接pass_”。

Boot 分区文件：[MEGA](https://mega.nz/file/wpJlUCIb#FyLieMpl03XE2nzdRr-XGgAw8tE537a1Y9qyyq-1QIc)

而文件中对应的 dtb 为 `gxl_p230_2g.dtb`。

不知 @ophub 老哥能否根据这些信息支持一些尚未支持的电视盒子？

ophub, 2023-06-27:
可以根据现有的信息测试相似的固件，以实测结果为准。

hzlyu123, 2023-06-29:
经过测试，与t95型号兼容，已经安装成功了：
openwrt：openwrt-06.26.2023-s905x-t95.img.gz
armbian：Armbian_23.08.0_amlogic_s905x-t95_jammy_5.15.117_server_2023.06.16.img.gz

hzlyu123, 2023-06-29:
感谢各位大佬回复！

closed 2023-06-29 by hzlyu123

ophub, 2023-06-29:
新魔百盒M201-S是s905l吧？把盒子的正面和反面拍个照片分享一下，底部能看清M201-S标识。
你测试下6.1.y内核固件能不能启动。

hzlyu123, 2023-06-29:
1.盒子正面和反面的照片参见附件。
2.新魔百盒M201-S是s905L，但是不知道对应哪个设备
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/make-openwrt/openwrt-files/common-files/etc/model_database.conf
；我是通过https://www.jianshu.com/p/3da8cb5e09a6这个网页找到对应的dtb，然后再从
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/make-openwrt/openwrt-files/common-files/etc/model_database.conf
找到对应设备的。
3.6.1y的内核估计启动报错，报错信息参见附件。

hzlyu123, 2023-06-29:
6.1.y用的固件是：openwrt_official_amlogic_s905x-t95_k6.1.34_2023.06.16.img.gz

hzlyu123, 2023-06-29:
用s905l的固件，5.15.y和6.1.y都可以。
5.5.y：openwrt_official_amlogic_s905l_k5.15.117_2023.06.22.img.gz
6.1.y：openwrt_official_amlogic_s905l_k6.1.36_2023.06.28.img.gz

ophub, 2023-06-29:
感谢分享，这个设备已经根据你的测试结果，添加到s905l设备列表里了。

https://github.com/ophub/amlogic-s9xxx-openwrt/commit/50a2e81b3f50e1b7f4b862824b1dfeea19b352eb

yjf007, 2023-08-25:
浙江移动魔百盒HM201刷哪个，有没有大佬成功刷入？
[image]

BailOrgana, 2023-09-10:
> 看了看，SoC 为 `Amlogic S905L`，目前没有明确支持 M201-S，没有 `dtb` 文件。
> 
> 你可以尝试刷入 [openwrt_lede_amlogic_s905l_k6.1.34_2023.06.22.img.gz](https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_lede_save_2023.06/openwrt_lede_amlogic_s905l_k6.1.34_2023.06.22.img.gz)。
> 
> 我找到了一些来自[人中日月](https://space.bilibili.com/97745521/)的用于 EmuELEC 的 dtb 文件：[device_trees](https://pan.emuelec.cn/Emuelec_cn/integrate/Emuelec%20S905L/device_trees)
> 
> 此外我还从 [EMUELEC-4G.img](https://pan.emuelec.cn/Emuelec_cn/integrate/Emuelec%20S905L/EMUELEC%203.3%204G/EMUELEC-4G.img) 提取了 Boot 分区文件，[据博主人中日月称](https://www.bilibili.com/read/cv4420521/)：“_S905/S912这次镜像是在S905L盒子上制作的，所以2018年上半年以前的S905L/M2的机器（包括但不限于：数码视讯Q7、部分数码视讯Q5、魔百盒HM201、百视通R3300L、咪咕MG101、魔百盒M201S/M201D、部分创维E900V21C）不用替换dtb直接pass_”。
> 
> 文件中对应的 dtb 为 `gxl_p230_2g.dtb`。
> 
> 不知 @ophub 老哥能否根据这些信息支持一些尚未支持的电视盒子？

> 看了看，SoC 为 ，目前没有明确支持 M201-S，没有 文件。`Amlogic S905L``dtb`
> 
> 你可以尝试刷入 [openwrt_lede_amlogic_s905l_k6.1.34_2023.06.22.img.gz](https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_lede_save_2023.06/openwrt_lede_amlogic_s905l_k6.1.34_2023.06.22.img.gz)。
> 
> 我找到了一些来自[人中日月](https://space.bilibili.com/97745521/)的用于 EmuELEC 的 dtb 文件：[device_trees](https://pan.emuelec.cn/Emuelec_cn/integrate/Emuelec%20S905L/device_trees)
> 
> 此外我还从 [EMUELEC-4G.img](https://pan.emuelec.cn/Emuelec_cn/integrate/Emuelec%20S905L/EMUELEC%203.3%204G/EMUELEC-4G.img) 提取了 Boot 分区文件，[据博主人中日月称](https://www.bilibili.com/read/cv4420521/)：“S905/S912这次镜像是在S905L_盒子上制作的，所以2018年上半年以前的S905L/M2的机器（包括但不限于：数码视讯Q7、部分数码视讯Q5、魔百盒HM201、百视通R3300L、咪咕MG101、魔百盒M201S/M201D、部分创维E900V21C）不用替换dtb直接pass_”。
> 
> 文件中对应的 dtb 为 。`gxl_p230_2g.dtb`
> 
> 不知 @ophub 老哥能否根据这些信息支持一些尚未支持的电视盒子？

大佬就是说这个固件[openwrt_lede_amlogic_s905l_k6.1.34_2023.06.22.img.gz](https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_lede_save_2023.06/openwrt_lede_amlogic_s905l_k6.1.34_2023.06.22.img.gz)。
能在HM201上直接用么

cherinyy, 2023-09-10:
> 浙江移动魔百盒HM201刷哪个，有没有大佬成功刷入？

> 能在HM201上直接用么

S905L-B 似乎可以通刷 S905L 的固件，你们也可以试试 [openwrt_lede_amlogic_s905l_k6.1.52_2023.09.08.img.gz](https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_lede_save_2023.09/openwrt_lede_amlogic_s905l_k6.1.52_2023.09.08.img.gz)，不一定行。

BailOrgana, 2023-09-10:
> > 浙江移动魔百盒HM201刷哪个，有没有大佬成功刷入？
> 
> > 能在HM201上直接用么
> 
> S905L-B 似乎可以通刷 S905L 的固件，你们也可以试试 [openwrt_lede_amlogic_s905l_k6.1.52_2023.09.08.img.gz](https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_lede_save_2023.09/openwrt_lede_amlogic_s905l_k6.1.52_2023.09.08.img.gz)，不一定行。

我发现使用的是第三方的盒子的系统，插上U盘，它不会自动进入U盘的系统引导界面。。。

niupipi, 2024-01-31:
> > 浙江移动魔百盒HM201刷哪个，有没有大佬成功刷入？
> 
> 
> 
> 
> 
> 
> 
> > 能在HM201上直接用么
> 
> 
> 
> S905L-B 似乎可以通刷 S905L 的固件，你们也可以试试 [openwrt_lede_amlogic_s905l_k6.1.52_2023.09.08.img.gz](https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_lede_save_2023.09/openwrt_lede_amlogic_s905l_k6.1.52_2023.09.08.img.gz)，不一定行。

报告一个测试结果。浙江HM201盒子使用Armbian_24.2.0_amlogic_s905lb-r3300l_jammy_6.1.74_server_2024.01.25优盘启动成功，armbian-install的时候，选0，进入手动配置:
Please Input SoC Name (such
as s9xxx): s905lb

Please Input DTB Name(such as meson-xxx.dtb): meson-gxl-s905x-p212.dtb

Please Input UBOOT_OVERLOAD Name(such as u-boot-xxx. bin): u-boot-p212.bin

Please Input MAINLINE_UBOOT Name (such as xxx-u-boot.bin.sd.bin) :(空白)

Please Input BOOTLOADER_IMG Name(such as xxx-bootloader. img):(空白)

hzlyu123, 2024-02-04:
浙江移动魔百盒HM201刷：openwrt_official_amlogic_s905l_k5.15.117_2023.06.22.img.gz

修改烧录好的U盘boot分区下的uEnv.txt文件中的dtb_name值为：
FDT=/dtb/amlogic/meson-gxl-s905x-nexbox-a95x.dtb

xiaoyuyouyiu, 2024-10-26:
大佬们，我有个魔百盒211-1增强版，2+16那个，芯片是905L的，用哪个可以刷

ilcicksie33332, 2024-12-09:
> 大佬们，我有个魔百盒211-1增强版，2+16那个，芯片是905L的，用哪个可以刷

我也在找，我找到一个，其他地方下载的。但是好像只能运行在U盘里面，我还在尝试其他固件