ophub/amlogic-s9xxx-openwrt #445: make: *** No rule to make target 'luci-app-amlogic/compile'.  Stop.

murad0522, 2023-06-20:
hello 

I'm trying to compile this app but I can't  solve this error 

`make: *** No rule to make target 'luci-app-amlogic/compile'.  Stop.`

I tried this down but no success 

```
# Add luci-app-amlogic
svn co https://github.com/ophub/luci-app-amlogic/trunk/luci-app-amlogic package/luci-app-amlogic

# This plugin can be compiled separately
make package/luci-app-amlogic/compile V=99

# Or integrate this plugin when fully compiling OpenWrt
make menuconfig
# choose LuCI ---> 3. Applications  ---> <*> luci-app-amlogic ----> save
make V=99
```

ophub, 2023-06-20:
```
# first:
git clone https://github.com/openwrt/openwrt openwrt
cd openwrt

# Add luci-app-amlogic
svn co https://github.com/ophub/luci-app-amlogic/trunk/luci-app-amlogic package/luci-app-amlogic

# This plugin can be compiled separately
make package/luci-app-amlogic/compile V=99

```

murad0522, 2023-06-20:
i can't find my chip in the target system when I try to compile it

ophub, 2023-06-21:
```
~/openwrt/bin/packages/aarch64_generic/base/

luci-app-amlogic_3.1.190-1_all.ipk
luci-i18n-amlogic-zh-cn_3.1.190-1_all.ipk
```

ophub, 2023-06-22:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/make-openwrt/documents

```
Target System  -> Arm SystemReady (EFI) compliant
Subtarget      -> 64-bit (armv8) machines
Target Profile -> Generic EFI Boot
Target Images  -> tar.gz

OR

Target System  -> QEMU ARM Virtual Machine
Subtarget      -> QEMU ARMv8 Virtual Machine (cortex-a53)
Target Profile -> Default
Target Images  -> tar.gz
```

murad0522, 2023-06-22:
`ake[2]: Leaving directory '/home/murad/test/ROOterSource2102/package/utils/e2fsprogs'
time: package/utils/e2fsprogs/compile#0.10#0.05#0.15
make[2]: Entering directory '/home/murad/test/ROOterSource2102/package/luci-app-amlogic'
/home/murad/test/ROOterSource2102/feeds/luci/luci.mk:316: warning: overriding recipe for target '/home/murad/test/ROOterSource2102/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/luci-app-amlogic.default.install.luci-i18n-amlogic-zh-cn'
/home/murad/test/ROOterSource2102/feeds/luci/luci.mk:316: warning: ignoring old recipe for target '/home/murad/test/ROOterSource2102/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/luci-app-amlogic.default.install.luci-i18n-amlogic-zh-cn'
/home/murad/test/ROOterSource2102/feeds/luci/luci.mk:316: warning: overriding recipe for target '/home/murad/test/ROOterSource2102/build_dir/target-aarch64_cortex-a53_musl/luci-app-amlogic/.pkgdir/luci-i18n-amlogic-zh-cn.installed'
/home/murad/test/ROOterSource2102/feeds/luci/luci.mk:316: warning: ignoring old recipe for target '/home/murad/test/ROOterSource2102/build_dir/target-aarch64_cortex-a53_musl/luci-app-amlogic/.pkgdir/luci-i18n-amlogic-zh-cn.installed'
/home/murad/test/ROOterSource2102/feeds/luci/luci.mk:316: warning: overriding recipe for target '/home/murad/test/ROOterSource2102/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/stamp/.luci-i18n-amlogic-zh-cn_installed'
/home/murad/test/ROOterSource2102/feeds/luci/luci.mk:316: warning: ignoring old recipe for target '/home/murad/test/ROOterSource2102/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/stamp/.luci-i18n-amlogic-zh-cn_installed'
/home/murad/test/ROOterSource2102/feeds/luci/luci.mk:316: warning: overriding recipe for target '/home/murad/test/ROOterSource2102/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/luci-i18n-amlogic-zh-cn.provides'
/home/murad/test/ROOterSource2102/feeds/luci/luci.mk:316: warning: ignoring old recipe for target '/home/murad/test/ROOterSource2102/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/luci-i18n-amlogic-zh-cn.provides'
/home/murad/test/ROOterSource2102/feeds/luci/luci.mk:316: warning: overriding recipe for target '/home/murad/test/ROOterSource2102/bin/packages/aarch64_cortex-a53/base/luci-i18n-amlogic-zh-cn_unknown_all.ipk'
/home/murad/test/ROOterSource2102/feeds/luci/luci.mk:316: warning: ignoring old recipe for target '/home/murad/test/ROOterSource2102/bin/packages/aarch64_cortex-a53/base/luci-i18n-amlogic-zh-cn_unknown_all.ipk'
/home/murad/test/ROOterSource2102/feeds/luci/luci.mk:316: warning: overriding recipe for target 'luci-i18n-amlogic-zh-cn-clean'
/home/murad/test/ROOterSource2102/feeds/luci/luci.mk:316: warning: ignoring old recipe for target 'luci-i18n-amlogic-zh-cn-clean'
make[2]: Nothing to be done for 'compile'.
make[2]: Leaving directory '/home/murad/test/ROOterSource2102/package/luci-app-amlogic'
time: package/luci-app-amlogic/compile#0.07#0.01#0.08
make[1]: Leaving directory '/home/murad/test/ROOterSource2102'
`
`WARNING: Makefile 'package/luci-app-amlogic/Makefile' has a dependency on 'parted', which does not exist`

I don't know where I'm wrong

ophub, 2023-06-22:
You directly `fork` the repo and click `Actions` to compile.

Remember to add `token`, the method is in the usage documentation of the repo.

murad0522, 2023-06-22:
* pkg_hash_check_unresolved: cannot find dependency luci-lib-fs for luci-app-amlogic
 * pkg_hash_check_unresolved: cannot find dependency parted for luci-app-amlogic
 * pkg_hash_check_unresolved: cannot find dependency resize2fs for luci-app-amlogic
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-amlogic found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-amlogic.

sorry for being annoying  , this is how far I got ,  
parted is installed and resize2fs

ophub, 2023-06-22:
Use the `config` in the repo, rename the config to .config and put it in the root directory of the openwrt source code

murad0522, 2023-06-22:
I managed to install it and build a new image and all went well and booted very well from sd, but now I got a new problem, when I try to install the system from sd to internal storage  the device boot till it gives this msg and keeps repeat it 

https://ibb.co/y04TC9w

ophub, 2023-06-23:
what’s your box?
use 5.15.y kernel

murad0522, 2023-06-23:
vontar x3
both kernel give same error. when installed to internal storage

i used your build nothing happened. I believe it my build fault but I don't know where the error

im gonna start fresh from zero.. I will install new os and new build.

ophub, 2023-06-23:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/make-openwrt/documents#1011-openwrt-required-options

look at 10.11

murad0522, 2023-06-23:
man you the best.. I always wonder what I should choose what I shouldn't.. this is really helpful

i will update later with results.. really thank you

murad0522, 2023-06-25:
still have the same problem, can't install it to internal storage
I noticed this at boot maybe the problem 

https://ibb.co/rmwhX3F

closed 2023-07-02 by ophub