ophub/amlogic-s9xxx-openwrt #446: Kernel v5 with dependent packages of Kernel v6

Tasmoto, 2023-06-24:
Hello,  I am managing Openwrt on an ATV box X96 mini (S905W).

My knowledge of Openwrt is average and I'm having Kernel version incompatibility problems when installing certain packages.

I use Kernel V5 because I can't use V6 on this device and I can't install certain packages because they depend on Kernel V6.

There is a place where I can download packages for this platform but that depend up to Kernel version 5? Any feed?

Thank you so much.

ophub, 2023-06-25:
Some kmods need to be integrated at compile time and cannot be installed later.

Tasmoto, 2023-06-25:
OK, thanks a lot. I knew this but figured there were old versions of packages somewhere that I could use.

I'm currently using Docker for services that I can't use because I can't install the necessary packages.

Thank you very much for your great work. Greetings

ophub, 2023-06-25:
The openwrt in docker depends on the kernel of the host machine. You can update the kernel of the host machine.

closed 2023-07-02 by ophub