ophub/amlogic-s9xxx-openwrt #447: 23.03.0-rc2

esaaprillia, 2023-06-28:
@ophub 

please add armsr already released

ophub, 2023-06-28:
rc1-rc5 is beta

esaaprillia, 2023-06-28:
> rc1-rc5 is beta

yes i know that but it needs to be added urgently. replace the armvirt target with armsr and it will be a different target and I see that the armsr target must be separated from armvirt

OwO-GitHub, 2023-06-28:
immortal fw4 装上之后 防火墙设置里不显示  fullcone nat1的选项

ophub, 2023-06-29:
https://github.com/ophub/amlogic-s9xxx-openwrt/commit/677d60db60523aadbed985cc2ad8a886f6e6c040

添加这个测试版选项了。

esaaprillia, 2023-06-29:
@ophub 
Thanks

esaaprillia, 2023-06-29:
@ophub 

```
[ STEPS ] Start adding custom files...
[ INFO ] [ files ] directory status: total 4
drwxr-xr-x 3 runner docker 4096 Jun 29 19:25 etc
[ STEPS ] Start building OpenWrt with Image Builder...
Profile "Default" does not exist!
Use "make info" to get a list of available profile names.
make[1]: *** [Makefile:236: _check_profile] Error 1
make: *** [Makefile:258: image] Error 2
```

esaaprillia, 2023-06-29:
maybe change make image PROFILE="Default" to make image PROFILE=""

esaaprillia, 2023-06-29:
there is no default profile in armsr anymore

esaaprillia, 2023-06-29:
maybe you made a change but you didn't try it. I already told you this must be separated armvirt and armsr

esaaprillia, 2023-06-29:
@ophub 

i saw your change and it solved the problem. i will close this

closed 2023-06-29 by esaaprillia

esaaprillia, 2023-07-12:
Ping @ophub 

this must be replaced

```sed -i "s|iw |ipconfig |g" ${wireless_mac80211}```

to

```sed -i 's|iw dev "$wdev" del |#iw dev "$wdev" del |g' ${wireless_mac80211}```

I've tried it doesn't change. can you change this