ophub/amlogic-s9xxx-openwrt #449: add mwan3 in S905x3 6.1.35-ophub  HK1-Box

n1019ws, 2023-06-29:
* pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-conncount found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.35-1-8b528299bfa24bfec9a42bd1ecb9eb8e) for kmod-ipt-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.35-1-8b528299bfa24bfec9a42bd1ecb9eb8e) for kmod-ipt-conntrack-extra
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-conntrack-extra found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.35-1-8b528299bfa24bfec9a42bd1ecb9eb8e) for kmod-ipt-ipopt
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-ipopt found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-mwan3:
 * 	kernel (= 6.1.35-1-8b528299bfa24bfec9a42bd1ecb9eb8e)
 * opkg_install_cmd: Cannot install package luci-app-mwan3.
 -----------
 Should I download the src to compile or how to fix the error "incompatible with the architectures configured"?

ophub, 2023-06-29:
`kmod` cannot be installed, it can only be added at compile time, you fork the warehouse, add the luci-app-xxx you need, and then compile

n1019ws, 2023-06-29:
Thanks so much for your great work. I just did:
vi /usr/lib/opkg/status
replace the fingerprint to [6.1.35-1-8b528299bfa24bfec9a42bd1ecb9eb8e]

closed 2023-06-29 by n1019ws