ophub/amlogic-s9xxx-openwrt #451: SmartLabs SML-5442TW

WallcroftUK, 2023-06-29:
[image]

is it possible to install it on this devices to emmc ?

ophub, 2023-06-29:
Send screenshots of the results of these three commands. There is currently no installation option for your device, you can add installation options according to your test situation.
```
cat /boot/uEnv.txt
cat /etc/flippy-openwrt-release
ls -l /boot/u-boot*
```

WallcroftUK, 2023-06-29:
[image]
[image]
[image]
 here

WallcroftUK, 2023-06-29:
i have SML-5442TW openwrt installed on sdcard i must edit uENV.txt to set my ddt

WallcroftUK, 2023-06-29:
meson-gxl-s905d-sml5442tw.dtb

WallcroftUK, 2023-06-29:
and if i can ask you guys about one thing 
if its possible to install k8s on device :D

WallcroftUK, 2023-06-29:
the thing is only this version of armbian work for me i tried latest armbian older armbian and nothing and after try of this img it actually booted :D and works

WallcroftUK, 2023-06-29:
and other thing wlan is not found no wifi even when this device have BT and Wifi

WallcroftUK, 2023-06-29:
I booted from sdcard and only working is your image of openwrt but if I wanted same way boot armbian it stuck at something

WallcroftUK, 2023-06-29:
Where should I search ? Sorry for that question

WallcroftUK, 2023-06-29:
found it :D

ophub, 2023-06-29:
Post the screenshot of the newly downloaded 6.1.36 interface after startup.
```
cat /etc/ophub-release
ls -l /boot/u-boot*
```

Installation method: armbian-install, select `104`

closed 2023-07-27 by ophub

07510480632, 2024-03-03:
cat /boot/uEnv.txt
cat /etc/flippy-openwrt-release
ls -l /boot/u-boot*

pospa1, 2024-06-09:
Hi, can someone help me with boot Armbian on SML-5442TW? Im not able to boot it from sd card... always boot into stock firmware.