ophub/amlogic-s9xxx-openwrt #452: cm211-1的网线无法成功连接

null12138, 2023-07-02:
**Device Information | 设备信息**
 - SOC: [S095x3]
 - Model [cm211-1]

**OpenWrt Version | 系统版本**
 - Kernel Version: [5.25]
 - Release: [F]

**Describe the bug | 问题描述**
之前刷的一个第三方openwrt，发现刷入emmc以后无法正常启动，后来换了一个f大编译的版本，可以成功写入，但是无法使用有线连接，报错进入混杂模式

Please provide a detailed description of the issue and, if necessary, attach a screenshot.
下面这个是可以连接的（第三方img
[image]
但是f大的是不可以连接到的
[image]
请您看看有头绪吗，万分感谢！

ophub, 2023-07-02:
这个cm211-1有不同的soc，我没见过你的这个s905x3版的。一般来说选择正确就可以使用

null12138, 2023-07-03:
对不起，刚开始我好像soc弄错了）
现在用了您编译的6内核，可以正常联网了，也可以正常进入网页了
但是宝盒刷入emmc后无法进入系统，TTL以后是这样的报错...麻烦您帮忙看下，谢谢！
[image]

ophub, 2023-07-05:
测试5.15.y系列的

null12138, 2023-07-05:
祝大佬生活愉快！

closed 2023-07-05 by null12138