ophub/amlogic-s9xxx-openwrt #454: DG-3399机型显示错误

LemonCrab666, 2023-07-05:
**Device Information | 设备信息**
 - SOC: [rk3399]
 - Model [DG-3399]

**OpenWrt Version | 系统版本**
 - Kernel Version: [6.1.37]
 - Release: [immortalwrt]

dg3399在openwrt的机型显示为 Device Model: Pine64 RockPro64 v2.1
晶晨宝盒也无法正确显示机器型号，无法使用功能更新内核

ophub, 2023-07-05:
没dts，使用的编译好的dtb，型号可以忽略，如果想显示自己喜欢的可以反编译修改，方法见文档。

不能更新内核的失败界面截图，提供详细操作描述。

LemonCrab666, 2023-07-07:
[rk3399-dg3399.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/11974446/rk3399-dg3399.zip)
我根据您的dtb反编译了更改了型号，您看能替换嘛，谢谢，内核更新是我操作错误，抱歉

ophub, 2023-07-07:
https://github.com/cm9vdA/build-linux/blob/master/boot/dts/rockchip/mainline/rk3399-dg3399.dts

我在这里找到了源码，我使用这个编译并更新了仓库。

LemonCrab666, 2023-07-07:
> https://github.com/cm9vdA/build-linux/blob/master/boot/dts/rockchip/mainline/rk3399-dg3399.dts
> 
> 我在这里找到了源码，我使用这个编译并更新了仓库。

好的，感谢

ophub, 2023-07-07:
源文件我提交到f大的6.1仓库了，以后的内核就使用这个文件了。

LemonCrab666, 2023-07-07:
> 源文件我提交到f大的6.1仓库了，以后的内核就使用这个文件了。

什么时候能更新呢，我重新打包op

ophub, 2023-07-07:
dtb已经更新了。https://github.com/ophub/amlogic-s9xxx-armbian/commit/6fd30e3fe8ab6c761d88b3cb0dbc0e26da8272c9

LemonCrab666, 2023-07-07:
> dtb已经更新了。[ophub/amlogic-s9xxx-armbian@6fd30e3](https://github.com/ophub/amlogic-s9xxx-armbian/commit/6fd30e3fe8ab6c761d88b3cb0dbc0e26da8272c9)

好的，感谢您的付出

closed 2023-07-07 by ophub