ophub/amlogic-s9xxx-openwrt #455: Panther X2 dtd file usb configuration is wrong

copyleft0, 2023-07-05:
**Device Information | 设备信息**
 - SOC: [rk3566]
 - Model [Panther X2]

**OpenWrt Version | 系统版本**
 - Kernel Version: [[k6.1.36]
 - Release: [official&mmortalwrt]

**Describe the bug | 问题描述**
存储插进去就断电重启，尤其是固态硬盘盒，并且重启后依然不识别。

查某群文件，
/boot/dtb/rockchip/rk3566-panther-x2.dtb
文件内，usb@fcc00000下，dr mode应为“host”，而非“otg”

[image]

从其他固件里抽来一个配置正确的dtd文件替换掉，故障就消失了

请核查并修改，谢谢！！

copyleft0, 2023-07-06:
[rk3566-panther-x2.dtb.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/11963469/rk3566-panther-x2.dtb.zip)
我只知道这个dtb的USB工作正常，其他参数是否一致，未核实。

另外还找到一份dts文件，我不会编译这玩意， 如需使用请先检查核实
[rk3566-panther-x2.dts.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/11963487/rk3566-panther-x2.dts.zip)
谢谢！

ophub, 2023-07-06:
dtb使用你分享的文件更新了。
dts不是源文件，是这个dtb反编译的文件，小改参数可以使用。

closed 2023-07-07 by ophub

copyleft0, 2023-07-07:
太好了

我现在还需要/能做什么嘛？
等待下一个版本更新之后试一下？

ophub, 2023-07-07:
我用的就是你的文件，你直接替换本地的文件即可。

ophub, 2023-07-07:
https://github.com/unifreq/linux-6.1.y/blob/main/arch/arm64/boot/dts/rockchip/rk3566-panther-x2.dts

我在armbian官方找到了这个源文件，现在添加到f大的内核仓库了。