ophub/amlogic-s9xxx-openwrt #456: Please add support for Akari AX512 S905x

hillz2, 2023-07-07:
Hi, please add support for akari AX512 box (s905x) I have the original android firmware (*.img) for this box if you need it. Let me know and I'll upload it for you.

[image]

ophub, 2023-07-07:
Use the existing s905x firmware to test

hillz2, 2023-07-07:
I have, and have tried several s905x dtbs, so far no dtb works for the box. Openwrt doesn't boot up

ophub, 2023-07-07:
try 5.15.y

zhz89512021, 2023-07-09:
It may be an accidental error during compilation, this image is still working properly.
https://github.com/zhz89512021/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_immortalwrt_save_2023.07

hillz2, 2023-07-09:
@zhz89512021 which kernel should I download for AX512?

zhz89512021, 2023-07-09:
[image]
6.1.38 on my device works will.

zhz89512021, 2023-07-09:
> @zhz89512021 which kernel should I download for AX512?

[image]
6.1.38 on my device works well

hillz2, 2023-07-09:
@zhz89512021 does everything else work like wifi and Ethernet?

zhz89512021, 2023-07-09:
> @zhz89512021 does everything else work like wifi and Ethernet?

it works well on my device
(my device doesn't have wifi,you can try it)

ophub, 2023-07-26:
@zhz89512021 

Hello, please share the configuration information you're using:

`cat /etc/flippy-openwrt-release`

closed 2023-07-26 by ophub

hillz2, 2023-07-28:
> @zhz89512021
> 
> Hello, please share the configuration information you're using:
> 
> `cat /etc/flippy-openwrt-release`

I have tried @zhz89512021 openwrt build for Akari AX512 but it still doesn't boot up, @zhz89512021 are you sure https://github.com/zhz89512021/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_immortalwrt_save_2023.07 is running well for Akari AX512 ? I'm referring to this box https://www.androidtv-guide.com/pay-tv-provider/xl-axiata-xl-home-box-dv8252/

allentd, 2024-01-21:
> I have tried @zhz89512021 openwrt build for Akari AX512 but it still doesn't boot up, @zhz89512021 are you sure [...]

Hi, how do we boot from SD card or USB on these official Android TV?