ophub/amlogic-s9xxx-openwrt #460: 江苏移动cm201-1 引导不进去。

zhouslin, 2023-07-27:
**Device Information | 设备信息**
 - SOC: [e.g. S095x3]
 - Model [e.g. HK1]

**OpenWrt Version | 系统版本**
 - Kernel Version: [e.g. 5.15.100]
 - Release: [e.g. official/lede/immortalwrt]

**Describe the bug | 问题描述**
All issues will only remain open for one week to prioritize resolving them.
After that period, they will be closed but can still continue to be discussed in the thread.
所有的问题都只保留一周的开启状态供重点关注解决，超时将关闭，但任然可以长期继续跟帖讨论。

Please provide a detailed description of the issue and, if necessary, attach a screenshot.
详细描述问题，并在必要时附上屏幕截图。

试过很多包都引导不进去。
[image]
[image]

cm201-1 江苏移动
每次都是这个界面然后重启。
这是试过的包
[image]
芯片信息
[image]
麻烦大神知道下。。

zhouslin, 2023-07-27:
感谢！！！后来也刷了9.0，看了很多issues才找到问题，原来是hdmi问题，屏幕就是不显示。其实已经引导进去了。可以ping通。
现在遇到的问题是安装openwrt后提示拔掉usb重启。重启不起来，屏幕也没有东西显示。明天先安装armbian试试吧。

ophub, 2023-07-27:
先试5.15的

zhouslin, 2023-07-28:
5.15试了也不行。开机的时候屏幕显示启动中。一会屏幕就熄了。

ophub, 2023-07-28:
启动1分钟后去路由器里看名字叫armbian的设备ip，使用ssh连接。
部分魔百盒没HDMI支持，黑屏但是已经正常启动了。

zhouslin, 2023-07-28:
装了armbian 也一样，安装成功拔掉重启不了。和openwrt区别是装了armbian 绿灯亮了。 openwrt一直红灯。
[image]
U盘引导进去是能看到12.221 开着shh。能连进去安装。
但是安装成功后重启，扫描不到开着ssh端口的ip。好像没有正常启动啊。

ophub, 2023-07-28:
有没有更换一个高版本的安卓9.0+系统

zhouslin, 2023-07-28:
安卓系统也是看到issues里别人分享的。
路由里面没有，一直显示红灯。 就插着u盘用吧。搞不定 - _-

zhouslin, 2023-07-28:
openwrt_lede_amlogic_s905l3-cm211_k5.15.121_2023.07.26.img
用了这个包可以了，感谢楼主！！

ophub, 2023-07-28:
好的👌

closed 2023-08-01 by ophub