ophub/amlogic-s9xxx-openwrt #462: 手头有一个深圳感臻的S905x2的电视盒子，U盘启动，跑码一段后没动静

SherlockGougou, 2023-08-02:
**Device Information | 设备信息**
 - SOC: [e.g. S095x2]
 - Model [e.g. SMB 256.06]

**OpenWrt Version | 系统版本**
 - Kernel Version: [e.g. 5.15.123]
 - Release: [e.g. openwrt_official_amlogic_s905x2_k5.15.123_2023.08.01]

**Describe the bug | 问题描述**
- U盘启动可以跑码，但是卡在下面的位置：
[image]

- 设备主板信息图片如下： 
[image]
[image]

s905x2的两个内核的镜像都测试了，通过网线连接到了上级路由器，但在上级路由器后台也没看到这个设备上线，应该是没启动成功。

SherlockGougou, 2023-08-02:
盒子原本系统是Android 9，内存2G，存储8G

SherlockGougou, 2023-08-02:
原系统内核信息如下：
[image]

SherlockGougou, 2023-08-02:
刚才测试了隔壁armbian的5.15.123内核的包，可以跑码完毕，到达这个页面，但是依旧没有连上网络，上级路由看不到设备：
[image]

SherlockGougou, 2023-08-02:
armbian安装成功：
[image]

ophub, 2023-08-02:
恭喜你，最终使用了什么配置？
cat /boot/uEnv.txt

SherlockGougou, 2023-08-02:
`armbian:jeff:# cat /boot/uEnv.txt
LINUX=/zImage
INITRD=/uInitrd
FDT=/dtb/amlogic/meson-g12a-x96-max-rmii.dtb
APPEND=root=UUID=53a1f59b-9ec8-4436-b782-e5cd2c6cb169 rootflags=data=writeback rw rootfstype=ext4 console=ttyAML0,115200n8 console=tty0 no_console_suspend consoleblank=0 fsck.fix=yes fsck.repair=yes net.ifnames=0 max_loop=128 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1`

但是我想安装的是openwrt呀 
[破涕为笑]

SherlockGougou, 2023-08-02:
我再试一下openwrt的镜像。

SherlockGougou, 2023-08-02:
好吧，我的锅！我在openwrt跑码停止后，插上键盘，按回车，结果就启动好了的。看来是跑码没刷新：
[image]

SherlockGougou, 2023-08-02:
此贴终结！感谢大佬！

closed 2023-08-02 by SherlockGougou

ophub, 2023-08-02:
openwrt跑码就是这样，跑不到banner那，要看得键盘回车。其实已经启动了，同网段流览器访问192.168.1.1即可

ophub, 2023-08-02:
你的盒子以前没见过，是新设备，我把他添加到支持的设备列表里。
把你的盒子拍几张照片分享下，外观正面照片，底部照片（把盒子型号的标志拍清楚点），如果已经拆机了把主板正面、反面照片也拍个，不方便拆机就不用了。

SherlockGougou, 2023-08-02:
是这样的，这个盒子是我们公司定制的盒子，在外面是买不到的。我是自己没事儿干尝试一下。

ophub, 2023-08-02:
好的，私有设备不用拍了