ophub/amlogic-s9xxx-openwrt #464: No /etc/config/network

Zas09, 2023-08-10:
**Device Information | 设备信息**
 - SOC: s922xj
 - Model Ugoos Am6 Plus

**OpenWrt Version | 系统版本**
 - Kernel Version: latest kernel 6.1
 - Release: lede latest build

Lan its not working and wireless also nothing.
[image]
missing /etc/config/network ... How to fix it?

ophub, 2023-08-10:
https://openwrt.org/docs/guide-user/base-system/basic-networking

https://openwrt.org/docs/guide-user/base-system/basic-networking

Zas09, 2023-08-11:
> In general, you don't need to create it manually, it will be created automatically at startup.
> 
> https://openwrt.org/docs/guide-user/base-system/basic-networking

So it's mean my box cant use that kernel?
i had try follow exactly like on that link ... Cant solve the problem ...
Then i try another kernel like kernel 5.4 and so on.. got lan working but dont have wireless ( Network>>Wireless
[image]
no wifi in this box 😭

ophub, 2023-08-11:
I see that you're currently using kernel version 5.10.189. If this kernel can function properly with wired connections, then the 6.1.y series should work as well. You can give it another try. Alternatively, you can directly update your kernel to the 6.1.y series. Navigate to System --- Amlogic Service --- Settings (select the 6.1.y series kernel) --- Online Download and Update. This will allow you to switch to the 6.1.y series.

What wireless chip model are you using? What driver is needed?

Zas09, 2023-08-11:
Thanks for answer my question and I think I will stay on this kernel.
I dont know how to check wireless chip of this model .. i'm newbies with this things,sorry for that . Also i unable to open this box .  , its solid/hard box to open to see the mainboard . Big SORRY but drop a photos of this box
[image]
[image]

ophub, 2023-08-12:
The drivers of the four series of kernels are slightly different. You can also try the firmware of other kernels.

Zas09, 2023-08-12:
> The drivers of the four series of kernels are slightly different. You can also try the firmware of other kernels.

I have try all these kernel 5.4.y ,5.10.y , 5.15.y , 6.1.y no wifi is working ..
I think i can close the issue .. just buy router for access point(wifi)

closed 2023-08-12 by Zas09

tri-pole, 2023-08-13:
I have an Am6 plus (but an earlier batch, I think, judging from the serial number). My box has an AMPAK AP6359S WiFi+BT chip. It works great with all Ophub 5.15.y, 6.1.y kernels, mainline kernels etc., using the combo (kernel) driver brcm4359, provided you have these firmware files

```
ls -l /lib/firmware/brcm/*4359* | cut -b 23-
  67687 Sep 24  2022 /lib/firmware/brcm/BCM4359C0.hcd
 576532 Sep 24  2022 /lib/firmware/brcm/brcmfmac4359-sdio.bin
   5888 Sep 24  2022 /lib/firmware/brcm/brcmfmac4359-sdio.txt
     21 Sep 29  2022 /lib/firmware/brcm/brcmfmac4359-sdio.ugoos,am6.bin -> brcmfmac4359-sdio.bin
     21 Sep 29  2022 /lib/firmware/brcm/brcmfmac4359-sdio.ugoos,am6.txt -> brcmfmac4359-sdio.txt

```

The file BCM4359C0.hcd is for Bluetooth and you probably have to hunt for it on the internet. The rest of the files are for WiFi and maybe you have to hunt for them too (they don't seem to be included in the package linux-firmware and I don't recall where I got them from). Don't forget to set up the symlinks as indicated above.

If you have a different WiFi+BT chip, try the Manjaro ARM forums for hints.

P.S. I use this dtb: meson-g12b-ugoos-am6.dtb

ophub, 2023-08-13:
Copy your driver files, compress them into a zip file and upload it, I will add it to the firmware

https://github.com/ophub/firmware

Zas09, 2023-08-14:
> I have an Am6 plus (but an earlier batch, I think, judging from the serial number). My box has an AMPAK AP6359S WiFi+BT chip. It works great with all Ophub 5.15.y, 6.1.y kernels, mainline kernels [...]

Hi, could you show me how to identify wifi chipset on ugoos am6 plus?

Zas09, 2023-08-14:
> ```
>  67687 Sep 24  2022 /lib/firmware/brcm/BCM4359C0.hcd
>  576532 Sep 24  2022 /lib/firmware/brcm/brcmfmac4359-sdio.bin
>    5888 Sep 24  2022 /lib/firmware/brcm/brcmfmac4359-sdio.txt
> ```
> 
> These files are included in the firmware, you can try to create a soft link according to the tri-pole method.

Yup these files include in the firmware, but the others no... Im about flash another firmware lede/official one 22.

tri-pole, 2023-08-14:
Unless you can get the driver to work with our proposed method, I don't know of any other way of identifying your chipset other than (1) internet search (e.g. Manjaro ARM forums), (2) booting some other distro that detects the chip correctly or (3) opening the box.

On my box it is necessary to set up the these two symlinks (otherwise the WiFi doesn't work):

```
21 Sep 29  2022 /lib/firmware/brcm/brcmfmac4359-sdio.ugoos,am6.bin -> brcmfmac4359-sdio.bin
     21 Sep 29  2022 /lib/firmware/brcm/brcmfmac4359-sdio.ugoos,am6.txt -> brcmfmac4359-sdio.txt
```

Zas09, 2023-08-15:
> 

root@OpenWrt:~# dmesg | grep -iE "(ampak|brcm|4359)"
[   18.668984] brcmfmac: F1 signature read @0x18000000=0x17294359
[   18.672456] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4359-sdio for chip BCM4359/9
[   19.957337] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[   24.744008] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[   24.778802] usbcore: registered new interface driver brcmfmac
I got these.

tri-pole, 2023-08-15:
You haven't indicated if you have set up the symlinks or not. They are necessary.

xyyx, 2025-03-05:
I got working WiFi on AM6 with ap6398s chip. Extracted firmware from Android. And seems dtb for am6 is broken. WiFi only work with default gt king pro dtb.

[image]