ophub/amlogic-s9xxx-openwrt #465: Unable to download and use mwan3 on H68K

RioPlay, 2023-08-13:
**Device Information | 设备信息**
 - SOC: [e.g. Rockchip RK3568]
 - Model [Hinlink/LinkStar H68K-1432  /  HINLINK OPC-H68K Board]

**OpenWrt Version | 系统版本**
 - Kernel Version: [6.1.42-ophub]
 - Firmware Version: [OpenWrt SNAPSHOT r0-98e3f82 / LuCI Master git-23.188.24247-537a1e4]
 - Release: [official]

**Describe the bug | 问题描述**
When using the software package manager I am unable to download mwan3 (A tool for Multi-WAN configurations)
所有的问题都只保留一周的开启状态供重点关注解决，超时将关闭，但任然可以长期继续跟帖讨论。

Please provide a detailed description of the issue and, if necessary, attach a screenshot.
详细描述问题，并在必要时附上屏幕截图。
Once I perform the action to start the download of luci-app-mwan3 it completes the base download for the LuCI GUI but then fails downloading the remaining dependencies. I will attach below the outputs / errors:

*Executing package manager*
> Installing mwan3 (2.11.7-1) to root...
Downloading https[]()://downloads.openwrt.org/snapshots/packages/aarch64_generic/packages/mwan3_2.11.7-1_all.ipk

*Errors*

> Collected errors:
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-nf-reject
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-nf-ipt
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-nf-log
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-ipt-core
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-nfnetlink
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-ipt-ipset
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-nf-reject6
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-nf-conntrack
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-nf-conntrack6
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-nf-nat
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-nf-log6
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-crypto->
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-crypto-crc32c
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-lib-crc32c
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-nft-core
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-nft-compat
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-nf-ipt6
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-ip6tables
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-nf-conncount
> * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-conncount found, but incompatible with the architectures configured
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-ipt-conntrack
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-ipt-conntrack-extra
kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-conntrack-extra found, but incompatible with the architectures configured
> * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697) for kmod-ipt-ipopt
> * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-ipopt found, but incompatible with the architectures configured
> * satisfy_dependencies_for: Cannot satisfy the following dependencies for mwan3:
> * 	kernel (= 6.1.44-1-ab9114223521d834a68aa04f1b40e697)
> * opkg_install_cmd: Cannot install package mwan3.

I have tried flashing the OpenWRT image afresh and attempting to install the package without doing updates to opkg as well as with updates. It appears that the package it is trying to grab is too new for the current running kernel which is 6.1.42.

I'll attach an image of my current OPKG Configuration:
[image]

I am still relatively new to the whole submitting a ticket and provided all needed details that may be needed.

Not sure if compiling the latest version to match mwan3 is the best way to go about it? Or if there is maybe a repository I'm not seeing to add to the OPKG configuration to grab the correct version.

Thank you to anybody and all that takes a look at this!

ophub, 2023-08-13:
Some Linux kernel modules (kmod) need to be added during the kernel compilation and cannot be installed later through module loading. This is because certain modules require close integration with the core parts of the kernel to function correctly.

You can fork the repository and add `luci-app-mwan3`.
The method is as follows(4.1.2): https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/documents

RioPlay, 2023-08-13:
I am getting stuck at 3. Fork Repository and Set GH_TOKEN in the tutorial. I don't believe the layout is exactly the same as it is listed in the tutorial.

Thanks in advance!

closed 2023-09-02 by ophub