ophub/amlogic-s9xxx-openwrt #469: OpenWrt 23.05.0 cannot be installed on eMMC

grndis, 2023-08-29:
[image]

**Device Information
 - SOC: S095x3
 - Model X96 Max+ 1Gb

**Armbian Version 
 - Kernel Version: 5.xx and 6.xx
 - Release: bookworm

**Describe the bug 
I have successfully installed OpenWrt 23.05.0-rc2 and OpenWrt 23.05.0-rc3 on an SD card, but I am unable to boot when installing on eMMC. I have tried various kernel versions (5.xx and 6.xx) from both stable and dev branches, but the outcome remains the same. However, I have not encountered any issues with OpenWrt 22.03.5. Currently, I am running OpenWrt 23.05.0-rc3, booting from the SD card. 

I believe it is important to find a solution for this problem, as it is the next stable release. Thank you.

ophub, 2023-08-29:
Tell me about your installation method and installation options. Did you choose to enable bootloader support. Whether the correct device number is selected against the device list.

grndis, 2023-08-29:
I create an OpenWrt image using the official builder with the following command:
```shell
make image PROFILE="generic" PACKAGES="\
acpid attr base-files bash bc blkid block-mount blockd bsdtar \
        btrfs-progs busybox bzip2 cgi-io chattr comgt comgt-ncm containerd coremark \
        coreutils coreutils-base64 coreutils-nohup coreutils-truncate curl coreutils-comm coreutils-od coreutils-stty \
        dosfstools dumpe2fs e2freefrag e2fsprogs exfat-mkfs irqbalance adguardhome \
        f2fs-tools f2fsck fdisk gawk getopt gzip hostapd-common iconv iw iwinfo jq jshn \
        kmod-brcmfmac kmod-brcmutil kmod-cfg80211 kmod-mac80211 libjson-script \
        liblucihttp liblucihttp-lua losetup lsattr lsblk lscpu mkf2fs nano htop \
        mount-utils openssl-util parted perl-http-date perlbase-file perlbase-getopt \
        perlbase-time perlbase-unicode perlbase-utf8 pigz ppp ppp-mod-pppoe \
        proto-bonding pv rename resize2fs runc subversion-client subversion-libs tar \
        tini ttyd tune2fs uclient-fetch uhttpd uhttpd-mod-ubus unzip uqmi usb-modeswitch \
        uuidgen wget-ssl whereis which wpad-basic wwan xfs-fsck xfs-mkfs xz \
        xz-utils ziptool zoneinfo-asia zoneinfo-core zstd \
        \
        luci luci-base luci-compat luci-lib-base luci-app-samba4 luci-app-sqm \
        luci-lib-ip luci-lib-ipkg luci-lib-jsonc luci-lib-nixio  \
        luci-mod-admin-full luci-mod-network luci-mod-status luci-mod-system  \
        luci-proto-3g luci-proto-bonding luci-proto-ipip luci-proto-ipv6 luci-proto-ncm  \
        luci-proto-openconnect luci-proto-ppp luci-proto-qmi luci-proto-relay luci-proto-wireguard unetd \
        \
        luci-app-amlogic zoneinfo-asia" FILES="files"
```         
     

then build from the Ophub repository using that image.

```shell
sudo ./make -b s905x3 -k 6.1.49 -s 2560 -u dev
```         

Install OpenWrt 23.05.0 on eMMC from Luci. However, it fails to boot. OpenWrt 22.03.5, on the other hand, has no booting issues. Additionally, OpenWrt 22.03.5 with kernel version 6.xx only works when using kernel dev.

ophub, 2023-08-29:
system > amlogic service > setting > select bootloader

install openwrt

grndis, 2023-08-29:
[image]
I tried it and the installation process seems successful as indicated by a message. However, the system fails to boot after removing the SD card. It can only boot again when the SD card is reinserted.

grndis, 2023-08-30:
[image]

I tested with another device (H96 Max X3), and encountered the same issue. It fails to boot when installed on eMMC. This is the firmware forked from ophub/amlogic-s9xxx-openwrt https://github.com/grndis/amlogic-s905x3-openwrt

ophub, 2023-08-30:
x96max(s905x3) and h96max(s905x3) have not found that they cannot be written to emmc. I don’t know what happened to you. Is it a new variant?

grndis, 2023-09-01:
After attempting multiple installations of various firmware and kernels to the eMMC, none of the openwrt firmwares can be successfully installed. Despite using a stable firmware provided by you, the device fails to boot after removing the SD card. However, I have managed to successfully install Armbian to the eMMC using the "armbian-install" method. Now, installing any OpenWrt to eMMC through Luci is not working, but I was able to successfully install Armbian to eMMC. I'm currently stuck. Do you have any suggestions for troubleshooting this issue? Thank you

ophub, 2023-09-01:
The contents of the /boot partition of the two systems are exactly the same. One works, and so does the other.

grndis, 2023-09-01:
> The contents of the /boot partition of the two systems are exactly the same. One works, and so does the other.

Is there an alternative method to install OpenWrt on eMMC without using Luci, similar to the 'armbian-install' method?

ophub, 2023-09-01:
openwrt-install-amlogic

grndis, 2023-09-01:
> openwrt-install-amlogic

Thank you! With that command, I found the problem. There is a hidden folder called `.Spotlight-V100` in the `/boot/` directory of macOS that was causing the error. It's resolved now. Thanks again!

closed 2023-09-01 by grndis