ophub/amlogic-s9xxx-openwrt #474: Request support wifi n bluetooth on s905w-X96w RTL8723 wifi bluetooth

khongpt, 2023-09-09:
**Device Information | 设备信息**
 - SOC: s905w
 - Model X96w
Box label X96W  has 1G RAM+ 8G Rom. **RTL8723** wifi + bluetooth support

**OpenWrt Version | ImmortalWrt SNAPSHOT r27500-8c90527a80
 - Kernel Version: [5.15.131-ophub]
 - Release: [immortalwrt]

**Describe the bug | 问题描述**

Wired network working fine. But fail to using wifi and bluetooth

 root@ImmortalWrt:~# dmesg | grep Bluetooth
```
[   24.394377] Bluetooth: Core ver 2.22
[   24.396714] Bluetooth: Starting self testing
[   24.418083] Bluetooth: ECDH test passed in 18677 usecs
[   24.426723] Bluetooth: SMP test passed in 6207 usecs
[   24.429115] Bluetooth: Finished self testing
[   24.437729] Bluetooth: HCI device and connection manager initialized
[   24.444027] Bluetooth: HCI socket layer initialized
[   24.448869] Bluetooth: L2CAP socket layer initialized
[   24.453930] Bluetooth: SCO socket layer initialized
[   24.462124] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   24.464372] Bluetooth: BNEP filters: protocol multicast
[   24.469330] Bluetooth: BNEP socket layer initialized
[   24.518461] Bluetooth: HCI UART driver ver 2.3
[   24.520882] Bluetooth: HCI UART protocol H4 registered
[   24.524784] Bluetooth: HCI UART protocol BCSP registered
[   24.530206] Bluetooth: HCI UART protocol LL registered
[   24.535127] Bluetooth: HCI UART protocol ATH3K registered
[   24.540607] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   24.547090] Bluetooth: HCI UART protocol Intel registered
[   24.552481] Bluetooth: HCI UART protocol Broadcom registered
[   24.557845] Bluetooth: HCI UART protocol QCA registered
[   24.572132] Bluetooth: HCI UART protocol AG6XX registered
[   24.590219] Bluetooth: HCI UART protocol Marvell registered
[   24.817157] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[   24.819914] Bluetooth: HIDP socket layer initialized
[   25.090954] Bluetooth: RFCOMM TTY layer initialized
[   25.093457] Bluetooth: RFCOMM socket layer initialized
[   25.097772] Bluetooth: RFCOMM ver 1.11
[   26.934825] Bluetooth: hci0: command 0xfc18 tx timeout
[   34.828033] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[   34.831067] Bluetooth: hci0: Failed to set baudrate
[   36.961405] Bluetooth: hci0: command 0x0c03 tx timeout
[   44.854747] Bluetooth: hci0: BCM: Reset failed (-110)
```

```
[   34.828033] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[   34.831067] Bluetooth: hci0: Failed to set baudrate
[   36.961405] Bluetooth: hci0: command 0x0c03 tx timeout
[   38.969496] meson8b-dwmac c9410000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   39.034393] meson8b-dwmac c9410000.ethernet eth0: PHY [0.e40908ff:08] driver [Meson GXL Internal PHY] (irq=52)
[   40.571505] meson8b-dwmac c9410000.ethernet eth0: No Safety Features support found
[   40.575898] meson8b-dwmac c9410000.ethernet eth0: PTP not supported by HW
[   40.582638] meson8b-dwmac c9410000.ethernet eth0: configuring for phy/rmii link mode
[   40.591690] br-lan: port 1(eth0) entered blocking state
[   40.595493] br-lan: port 1(eth0) entered disabled state
[   40.601083] device eth0 entered promiscuous mode
[   40.765260] rtl8723bs mmc0:0001:1 wlan0: rtw_ndev_notifier_call(wlan0) state:10
[   40.769971] rtl8723bs mmc0:0001:1 wlan0: rtw_ndev_notifier_call(wlan0) state:2
[   40.777867] rtl8723bs mmc0:0001:1 wlan0 (unregistering): rtw_ndev_notifier_call(wlan0) state:6
[   41.483052] NET: Registered PF_QIPCRTR protocol family
[   42.240230] meson8b-dwmac c9410000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   42.246029] br-lan: port 1(eth0) entered blocking state
[   42.250932] br-lan: port 1(eth0) entered forwarding state
[   42.256664] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   44.854747] Bluetooth: hci0: BCM: Reset failed (-110)
[   46.924494] BTRFS: device label USB_ROOTFS2 devid 1 transid 14 /dev/sda3 scanned by block (6831)
[   46.933106] BTRFS info (device sda3): using crc32c (crc32c-generic) checksum algorithm
[   46.938394] BTRFS info (device sda3): disk space caching is enabled
[   46.944573] BTRFS info (device sda3): has skinny extents
[   47.205773] BTRFS: device label USB_SHARED devid 1 transid 14 /dev/sda4 scanned by block (6858)
[   47.215020] BTRFS info (device sda4): using crc32c (crc32c-generic) checksum algorithm
[   47.220379] BTRFS info (device sda4): disk space caching is enabled
[   47.226540] BTRFS info (device sda4): has skinny extents
```

Please take alook how to support wifi and bluetooth for this box.

Thanks

ophub, 2023-09-09:
https://github.com/unifreq/linux-6.1.y/tree/main/drivers/net/wireless/realtek/rtlwifi

Check if there is a driver for your chip in the mainline kernel.

khongpt, 2023-09-10:
I tried to make manually with:

**sudo ./make -b s905w**
```
[ STEPS ] Welcome to make OpenWrt!
[ INFO ] Server running on Ubuntu: [ Release: focal / Host: x86_64 ]

[ STEPS ] Start Initializing Variables...
[ STEPS ] Start searching for OpenWrt file...
[ INFO ] OpenWrt file: [ openwrt-armsr-armv8-generic-rootfs.tar.gz ]
[ INFO ] The source_codename: [ _official ], OpenWrt source code repository: [ github.com/openwrt/openwrt ]
[ STEPS ] Start downloading dependency files...
[ INFO ] platform-files download completed.
[ INFO ] different-files download completed.
[ INFO ] u-boot download completed.
[ INFO ] firmware download completed.
A    make-openwrt/openwrt-files/common-files/usr/sbin/balethirq.pl
Export complete.
A    make-openwrt/openwrt-files/common-files/etc/balance_irq
Export complete.
[ INFO ] balethirq download completed.
[ INFO ] app/sbin download completed.
[ INFO ] app/share download completed.
[ STEPS ] Start querying the latest kernel version...
[ INFO ] (1.1) Auto query the latest kernel version for [ stable - 6.1.1 ]
[ INFO ] (1.1) [ stable - 6.1.52 ] is latest kernel.

[ INFO ] (1.2) Auto query the latest kernel version for [ stable - 5.15.1 ]
[ INFO ] (1.2) [ stable - 5.15.131 ] is latest kernel.

[ STEPS ] Start downloading the kernel files...
[ INFO ] (1.1) [ stable - 6.1.52 ] Kernel is in the local directory.
[ INFO ] All [ 6 ] kernel files are sha256sum checked to be complete.

[ INFO ] (1.2) [ stable - 5.15.131 ] Kernel is in the local directory.
[ INFO ] All [ 6 ] kernel files are sha256sum checked to be complete.

[ INFO ] [ 1 ] lists of OpenWrt board: [ s905w ]
[ INFO ] Kernel Repo: [ ophub/kernel ], Kernel Usage: [  ]

[ INFO ] Server space usage before starting to compile:
Filesystem     Type  Size  Used Avail Use% Mounted on
/dev/sda7      ext4   84G   69G   12G  87% /

[ STEPS ] Start making OpenWrt firmware...
(1.1) Start making OpenWrt [ s905w - stable/6.1.52 ]. Remaining space is 11G.
 [🌿] (1/6) Make OpenWrt image.
 [🌿] (2/6) Extract OpenWrt files.
 [🌿] (3/6) Replace the kernel.
 [🌿] (4/6) Refactor bootfs files.
 [🌿] (5/6) Refactor rootfs files.
 [🌿] (6/6) Cleanup tmp files.
(1.1) OpenWrt made successfully.

(1.2) Start making OpenWrt [ s905w - stable/5.15.131 ]. Remaining space is 10G.
 [🌿] (1/6) Make OpenWrt image.
 [🌿] (2/6) Extract OpenWrt files.
 [🌿] (3/6) Replace the kernel.
 [🌿] (4/6) Refactor bootfs files.
 [🌿] (5/6) Refactor rootfs files.
 [🌿] (6/6) Cleanup tmp files.
(1.2) OpenWrt made successfully.

[ STEPS ] Server space usage after compilation:
Filesystem     Type  Size  Used Avail Use% Mounted on
/dev/sda7      ext4   84G   69G   11G  87% /

[ SUCCESS ] All process completed successfully.
```

Box boot successfully with kernel version 5.15.131 Kernel, but not with kernel version 6.1.52. Can you tell me how they differ?
I've created a local OpenWrt build on my machine. Can you please tell me how to boot it with kernel 6.1.y? Thanks

ophub, 2023-09-10:
`sudo ./make -b s905w -u dev`

Any version of the dev kernel can be used, while stable requires u-boot debugging

closed 2023-09-28 by ophub