ophub/amlogic-s9xxx-openwrt #476: Failed to update the kernel online through LuCi.

grndis, 2023-09-18:
**Device Information | 设备信息**
 - SOC: S095x3
 - Model x96 max+

**OpenWrt Version | 系统版本**
 - Kernel Version: 6.1.49-ophub
 - Release: OpenWrt 23.05.0-rc3

**Describe the bug | 问题描述**

I encountered an error while attempting to update the kernel online via LuCi. The error message states: "[Error] Missing [tar] in OpenWrt firmware, unable to update kernel." 

I tried uploading manually, but I got the same error message.

Got any clues to help fix this problem?

Thank you.

[image]

ophub, 2023-09-18:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/documents

10.11

When compiling by yourself, all required options must be added.

grndis, 2023-09-18:
very clear, thank you 🤝🏼

closed 2023-09-18 by grndis