ophub/amlogic-s9xxx-openwrt #480: failed

itremote, 2023-09-22:
**Device Information | 设备信息**
 - SOC: S905X3
 - Model H96Max

**OpenWrt Version | 系统版本**
 - Kernel Version: [e.g. 5.15.100]
 - Release: official 22.03.5

**Describe the bug | 问题描述**
I have failed image builder build workflow with only h96 max option 

[image]

OldCoding, 2023-09-22:
[image]
https://github.com/itremote/amlogic-s9xxx-openwrt/blob/23ef6697631ea80e0bd01a462cede8dd2ce2ea87/.github/workflows/build-openwrt-with-imagebuilder.yml#L241
patch？

itremote, 2023-09-22:
> https://github.com/itremote/amlogic-s9xxx-openwrt/actions/runs/6270050206/job/17027414162#step:4:1320

thx.. just remove samba4 from config.. :)

closed 2023-09-22 by itremote