ophub/amlogic-s9xxx-openwrt #484: x96 air 100MB can't boot openwrt (10/2023) from sd card

HokaDiep, 2023-10-05:
**Device Information | 设备信息**
 - SOC: [e.g. S095x3]: x905x3
 - Model [e.g. HK1] : x96 air 100MB

**OpenWrt Version | 系统版本**
 - Kernel Version: [e.g. 5.15.100]: 5.15.133
 - Release: [e.g. official/lede/immortalwrt]: official

**Describe the bug | 问题描述**
Hi guys, I was able to successfully install armbian onto emmc, but I can't boot openwrt from SD card

Please provide a detailed description of the issue and, if necessary, attach a screenshot.
详细描述问题，并在必要时附上屏幕截图。

change dtb
[image]
[image]

I didn't change anything else
[image]
[image]

ophub, 2023-10-06:
From the system startup requirements perspective, the startup requirements for the OpenWRT downloaded from this repository are the same as Armbian. The dtb file name in uEnv.txt is the same, and firmware using the 5.15.y kernel can be used to start. Some specific boxes may be picky about USB, so trying a different USB may help. If you can replace the capacitor next to the USB interface on the motherboard with a solid-state capacitor, it will solve this problem.

HokaDiep, 2023-10-06:
> Từ góc độ yêu cầu khởi động hệ thống, các yêu cầu khởi động cho OpenWRT được tải xuống từ kho lưu trữ này giống như Armbian. Tên tệp dtb trong uEnv.txt giống nhau và chương trình cơ sở sử dụng kernel 5.15.y có thể được sử dụng để khởi động. Một số hộp cụ thể có thể kén chọn USB, vì vậy, việc thử một USB khác có thể hữu ích. Nếu bạn có thể thay thế tụ điện bên cạnh giao diện USB trên bo mạch chủ bằng tụ điện thể rắn thì sẽ giải quyết được vấn đề này.

Thanks for your reply. I have boot it successfully from sd card. Could you please tell me what command should I use if I change the kernel in ssh?

ophub, 2023-10-06:
The usage of common commands like updating the kernel is explained on the homepage, and more detailed instructions can be found in the documentation.

closed 2023-10-08 by ophub