ophub/amlogic-s9xxx-openwrt #488: HG680LC刷的固件貌似不支持swap

Xuchengen, 2023-10-11:
该固件貌似不支持swap
``block: failed to swapon swapfile (-1)``

另外：你们是否有计划支持WiFi。

ophub, 2023-10-11:
armbian-swap 1

Xuchengen, 2023-10-12:
## 内核
```
root@HG680LC:~# uname -a
Linux HG680LC 5.15.133-ophub #1 SMP PREEMPT Sat Sep 23 06:50:41 EDT 2023 aarch64 GNU/Linux
```

## 创建swap
```
root@HG680LC:~# openwrt-swap 1
[ STEPS ] Welcome to use swap management tool.
[ INFO ] Prepare to create [ 1 ] GiB swap in the [ /mnt/mmcblk2p4/.swap ] directory.
[ STEPS ] Start creating a new swap...
[ INFO ] The swap size is [ 1 ] GiB, partition types is [ btrfs ], Please wait...
/usr/sbin/openwrt-swap: line 87: truncate: command not found
chattr: No such file or directory while trying to stat /mnt/mmcblk2p4/.swap/swapfile
ERROR: failed to open /mnt/mmcblk2p4/.swap/swapfile: No such file or directory
ERROR: failed to detect object type: No such file or directory
1048576+0 records in
1048576+0 records out
Setting up swapspace version 1, size = 1073737728 bytes
block: failed to swapon /mnt/mmcblk2p4/.swap/swapfile (-1)
cfg052640
[ INFO ] Current swap: 
Filename				Type		Size		Used		Priority
[ SUCCESS ] Created successfully!
root@HG680LC:~# openwrt-swap 1
[ STEPS ] Welcome to use swap management tool.
[ INFO ] Prepare to create [ 1 ] GiB swap in the [ /mnt/mmcblk2p4/.swap ] directory.
[ STEPS ] Start creating a new swap...
[ INFO ] The swap size is [ 1 ] GiB, partition types is [ btrfs ], Please wait...
/usr/sbin/openwrt-swap: line 87: truncate: command not found
1048576+0 records in
1048576+0 records out
Setting up swapspace version 1, size = 1073737728 bytes
block: failed to swapon /mnt/mmcblk2p4/.swap/swapfile (-1)
cfg052640
[ INFO ] Current swap: 
Filename				Type		Size		Used		Priority
[ SUCCESS ] Created successfully!
```

ophub, 2023-10-12:
truncate: command not found
软件中心 -- 在线更新 -- 搜索安装

Xuchengen, 2023-10-15:
估计是磁盘btrfs文件系统不支持

closed 2023-10-15 by Xuchengen