ophub/amlogic-s9xxx-openwrt #489: 菜单里的终端不能用

goldsoft8888, 2023-10-14:
OpenWrt_official_save_2023.10
菜单里的终端不能用，TTYD

ophub, 2023-10-14:
改了IP要重启才能使用。

goldsoft8888, 2023-10-14:
> 修改了IP要重启才能使用。
重启无数次了，卸了重装都不行

ophub, 2023-10-14:
截图看看什么现象

goldsoft8888, 2023-10-14:
还有，想装WIREGUARD，提示内核版本不对。

goldsoft8888, 2023-10-14:
[image]

ophub, 2023-10-14:
[image]
[image]

goldsoft8888, 2023-10-14:
[image]
我是TX3的盒子，怎么都不行，前几天的版本，终端可以用的，升级到最新版的，就不行了，

closed 2023-11-05 by ophub