ophub/amlogic-s9xxx-openwrt #491: 希望能够编译支持 firefly rk3399 的openwrt

liaoya, 2023-10-18:
**Device Information | 设备信息**
 - SOC: [rk3399]
 - Model [firefly rk3399]

我尝试着使用 `openwrt_official_rockchip_tvi3315a_k6.1.57_2023.10.13.img` 写入到 tf 卡上，然后挂载 `/boot` 分区修改 `armbianEnv.txt` 将 `fdtfile=rockchip/rk3399-tvi3315a.dtb` 改为 `fdtfile=rockchip/rk3399-firefly.dtb`。用此 tf 卡无法成功启动系统在 UART 下看到如下内容

```
DDR Version 1.26 20210628
In
Channel 0: DDR3, 800MHz
Bus Width=32 Col=10 Bank=8 Row=15 CS=1 Die Bus-Width=16 Size=1024MB
Channel 1: DDR3, 800MHz
Bus Width=32 Col=10 Bank=8 Row=15 CS=1 Die Bus-Width=16 Size=1024MB
256B stride
ch 0 ddrconfig = 0x101, ddrsize = 0x20
ch 1 ddrconfig = 0x101, ddrsize = 0x20
pmugrf_os_reg[2] = 0x32817281, stride = 0x9
OUT
Boot1 Release Time: May 29 2020 17:36:36, version: 1.26
CPUId = 0x0
ChipType = 0x10, 337
mmc: ERROR: SDHCI:Transfer data timeout
mmc: ERROR: SDHCI ERR:cmd:0x153a,stat:0x0
mmc: ERROR: Tuning procedure failed, falling back to fixed sampling clock
mmc: ERROR: tuning execution failed
emmc reinit
mmc: ERROR: SDHCI:Transfer data timeout
mmc: ERROR: SDHCI ERR:cmd:0x153a,stat:0x0
mmc: ERROR: Tuning procedure failed, falling back to fixed sampling clock
mmc: ERROR: tuning execution failed
emmc reinit
mmc: ERROR: SDHCI:Transfer data timeout
mmc: ERROR: SDHCI ERR:cmd:0x153a,stat:0x0
mmc: ERROR: Tuning procedure failed, falling back to fixed sampling clock
mmc: ERROR: tuning execution failed
SdmmcInit=2 1
mmc0:cmd5,20
SdmmcInit=0 0
BootCapSize=0
UserCapSize=29862MB
FwPartOffset=2000 , 0
StorageInit ok = 35731907
SecureMode = 0
SecureInit read PBA: 0x4
SecureInit read PBA: 0x404
SecureInit read PBA: 0x804
SecureInit read PBA: 0xc04
SecureInit read PBA: 0x1004
SecureInit read PBA: 0x1404
SecureInit read PBA: 0x1804
SecureInit read PBA: 0x1c04
SecureInit ret = 0, SecureMode = 0
atags_set_bootdev: ret:(0)
GPT part:  0, name:          primary, start:0x8000, size:0x7f800
GPT part:  1, name:          primary, start:0x88000, size:0x1ff800
no find partition:uboot.
Trust Addr:0x4000, 0x58334c42
No find bl30.bin
No find bl32.bin
Load uboot, ReadLba = 2000
Load OK, addr=0x200000, size=0xb08f0
RunBL31 0x40000 @ 51981 us
NOTICE:  BL31: v1.3(debug):42583b6
NOTICE:  BL31: Built : 07:55:13, Oct 15 2019
NOTICE:  BL31: Rockchip release version: v1.1
INFO:    GICv3 with legacy support detected. ARM GICV3 driver initialized in EL3
INFO:    Using opteed sec cpu_context!
INFO:    boot cpu mask: 0
INFO:    plat_rockchip_pmu_init(1190): pd status 3e
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

U-Boot 2020.10-armbian (Aug 08 2021 - 16:19:56 +0200)

SoC: Rockchip rk3399
Reset cause: POR
Model: Orange Pi RK3399 Board
DRAM:  2 GiB
PMIC:  RK808
MMC:   mmc@fe310000: 2, mmc@fe320000: 1, sdhci@fe330000: 0
Loading Environment from MMC... *** Warning - !read failed, using default environment

In:    serial@ff1a0000
Out:   serial@ff1a0000
Err:   serial@ff1a0000
Model: Orange Pi RK3399 Board
```

我的这块板子有一些问题，已经无法识别内置的 EMMC。但是用来测试 tf 卡上的 OpenWRT 应该问题不大。我可以在 tf 卡上跑官方的 armbian.

ophub, 2023-10-18:
可以使用 rk3399-tvi3315a.dtb 的固件？只是名字的显示？

liaoya, 2023-10-18:
不仅是名字现实不一样，无法正常引导完成。

Armbian 是官方的 <https://redirect.armbian.com/region/AS/firefly-rk3399/Jammy_current>，使用的镜像的下载地址 <http://mirrors.ustc.edu.cn/armbian-dl/firefly-rk3399/archive/Armbian_23.8.1_Firefly-rk3399_jammy_current_6.1.50.img.xz>

ophub, 2023-10-18:
https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_official_save_2023.10/openwrt_official_rockchip_firefly_k6.1.58_2023.10.18.img.gz

https://github.com/ophub/amlogic-s9xxx-armbian/releases/download/Armbian_jammy_save_2023.10/Armbian_23.11.0_rockchip_firefly_jammy_6.1.58_server_2023.10.18.img.gz

这2个系统都帮忙测试下，看看是否都可以正常启动。

liaoya, 2023-10-18:
@ophub 非常感谢您的努力。我测试了两个镜像文件。型号显示已经正常了，但是系统仍然无法正常引导（我还接了一个显示器上去进行了确认）。日志如下

```
SdmmcInit=2 1
mmc0:cmd5,20
SdmmcInit=0 0
BootCapSize=0
UserCapSize=29862MB
FwPartOffset=2000 , 0
StorageInit ok = 35741920
SecureMode = 0
SecureInit read PBA: 0x4
SecureInit read PBA: 0x404
SecureInit read PBA: 0x804
SecureInit read PBA: 0xc04
SecureInit read PBA: 0x1004
SecureInit read PBA: 0x1404
SecureInit read PBA: 0x1804
SecureInit read PBA: 0x1c04
SecureInit ret = 0, SecureMode = 0
atags_set_bootdev: ret:(0)
GPT part:  0, name:          primary, start:0x8000, size:0x7f800
GPT part:  1, name:          primary, start:0x88000, size:0x4ff800
no find partition:uboot.
Trust Addr:0x4000, 0x58334c42
No find bl30.bin
No find bl32.bin
Load uboot, ReadLba = 2000
Load OK, addr=0x200000, size=0xd9f08
RunBL31 0x40000 @ 77286 us
NOTICE:  BL31: v1.3(release):845ee93
NOTICE:  BL31: Built : 15:51:11, Jul 22 2020
NOTICE:  BL31: Rockchip release version: v1.1
INFO:    GICv3 with legacy support detected. ARM GICV3 driver initialized in EL3
INFO:    Using opteed sec cpu_context!
INFO:    boot cpu mask: 0
INFO:    plat_rockchip_pmu_init(1196): pd status 3e
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

U-Boot 2022.04-armbian (Aug 22 2023 - 15:24:21 +0000)

SoC: Rockchip rk3399
Reset cause: POR
Model: Firefly-RK3399 Board
DRAM:  2 GiB
PMIC:  RK808
Core:  286 devices, 28 uclasses, devicetree: separate
MMC:   mmc@fe310000: 3, mmc@fe320000: 1, mmc@fe330000: 0
Loading Environment from MMC... *** Warning - bad CRC, using default environment

In:    serial
Out:   vidconsole
Err:   vidconsole
Model: Firefly-RK3399 Board
Net:   eth0: ethernet@fe300000
Hit any key to stop autoboot:  0
starting USB...
Bus usb@fe380000: USB EHCI 1.00
Bus usb@fe3c0000: USB EHCI 1.00
Bus usb@fe900000: Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.10
scanning bus usb@fe380000 for devices... 1 USB Device(s) found
scanning bus usb@fe3c0000 for devices... 2 USB Device(s) found
scanning bus usb@fe900000 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found

Device 0: unknown device
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
Found U-Boot script /boot.scr
3252 bytes read in 5 ms (634.8 KiB/s)
## Executing script at 00500000
Boot script loaded from mmc 1
268 bytes read in 4 ms (65.4 KiB/s)
16866283 bytes read in 720 ms (22.3 MiB/s)
33923584 bytes read in 1444 ms (22.4 MiB/s)
82844 bytes read in 12 ms (6.6 MiB/s)
Failed to load '/dtb/rockchip/overlay/rk3399-fixup.scr'
Unknown command 'kaslrseed' - try 'help'
Moving Image from 0x2080000 to 0x2200000, end=4340000
## Loading init Ramdisk from Legacy Image at 06000000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    16866219 Bytes = 16.1 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 01f00000
   Booting using the fdt blob at 0x1f00000
   Loading Ramdisk to 78ef6000, end 79f0bbab ... OK
   Loading Device Tree to 0000000078e79000, end 0000000078ef5fff ... OK

Starting kernel ...
```

不知道是不是和如下错误有关

```
Failed to load '/dtb/rockchip/overlay/rk3399-fixup.scr'
Unknown command 'kaslrseed' - try 'help'
```

ophub, 2023-10-18:
你先测试我的默认固件，不要添加其他

ophub, 2023-10-18:
两个文件重新下载一下再试，我改了一下

liaoya, 2023-10-18:
两个 image 我都测试了，这次 kernel 开始引导了，不过在自检那里停住了。然后显示器花屏以后就黑屏了

```
SdmmcInit=2 1
mmc0:cmd5,20
SdmmcInit=0 0
BootCapSize=0
UserCapSize=29862MB
FwPartOffset=2000 , 0
StorageInit ok = 35728093
SecureMode = 0
SecureInit read PBA: 0x4
SecureInit read PBA: 0x404
SecureInit read PBA: 0x804
SecureInit read PBA: 0xc04
SecureInit read PBA: 0x1004
SecureInit read PBA: 0x1404
SecureInit read PBA: 0x1804
SecureInit read PBA: 0x1c04
SecureInit ret = 0, SecureMode = 0
atags_set_bootdev: ret:(0)
GPT part:  0, name:          primary, start:0x8000, size:0x7f800
GPT part:  1, name:          primary, start:0x88000, size:0x1ff800
no find partition:uboot.
Trust Addr:0x4000, 0x58334c42
No find bl30.bin
No find bl32.bin
Load uboot, ReadLba = 2000
Load OK, addr=0x200000, size=0xd9f08
RunBL31 0x40000 @ 62913 us
NOTICE:  BL31: v1.3(release):845ee93
NOTICE:  BL31: Built : 15:51:11, Jul 22 2020
NOTICE:  BL31: Rockchip release version: v1.1
INFO:    GICv3 with legacy support detected. ARM GICV3 driver initialized in EL3
INFO:    Using opteed sec cpu_context!
INFO:    boot cpu mask: 0
INFO:    plat_rockchip_pmu_init(1196): pd status 3e
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

U-Boot 2022.04-armbian (Aug 22 2023 - 15:24:21 +0000)

SoC: Rockchip rk3399
Reset cause: POR
Model: Firefly-RK3399 Board
DRAM:  2 GiB
PMIC:  RK808
Core:  286 devices, 28 uclasses, devicetree: separate
MMC:   mmc@fe310000: 3, mmc@fe320000: 1, mmc@fe330000: 0
Loading Environment from MMC... *** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Model: Firefly-RK3399 Board
Net:   eth0: ethernet@fe300000
Hit any key to stop autoboot:  0
starting USB...
Bus usb@fe380000: USB EHCI 1.00
Bus usb@fe3c0000: USB EHCI 1.00
Bus usb@fe900000: Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.10
scanning bus usb@fe380000 for devices... 1 USB Device(s) found
scanning bus usb@fe3c0000 for devices... 2 USB Device(s) found
scanning bus usb@fe900000 for devices... 1 USB Device(s) found
[... 278 lines cut ...]
[    1.742227] usbserial: USB Serial support registered for ch341-uart
[    1.742815] usbcore: registered new interface driver cp210x
[    1.743330] usbserial: USB Serial support registered for cp210x
[    1.743896] usbcore: registered new interface driver ftdi_sio
[    1.744429] usbserial: USB Serial support registered for FTDI USB Serial Device
[    1.745105] usbcore: registered new interface driver garmin_gps
[    1.745655] usbserial: USB Serial support registered for Garmin GPS usb/tty
[    1.746305] usbcore: registered new interface driver ipw
[    1.746797] usbserial: USB Serial support registered for IPWireless converter
[    1.747457] usbcore: registered new interface driver opticon
[    1.747991] usbserial: USB Serial support registered for opticon
[    1.748554] usbcore: registered new interface driver pl2303
[    1.749069] usbserial: USB Serial support registered for pl2303
[    1.749623] usbcore: registered new interface driver qcaux
[    1.750129] usbserial: USB Serial support registered for qcaux
[    1.750680] usbcore: registered new interface driver qcserial
[    1.751215] usbserial: USB Serial support registered for Qualcomm USB modem
[    1.751875] usbcore: registered new interface driver symbolserial
[    1.752437] usbserial: USB Serial support registered for symbol
[    1.753586] mousedev: PS/2 mouse device common for all mice
[    1.755273] SPI driver rmi4_spi has no spi_device_id for syna,rmi4-spi
[    1.756759] i2c_dev: i2c /dev entries driver
[    1.759954] i2c 0-001b: Fixed dependency cycle(s) with /i2c@ff3c0000/pmic@1b/regulators/LDO_REG3
[    1.761558] fan53555-regulator 0-0040: FAN53555 Option[8] Rev[1] Detected!
```

ophub, 2023-10-18:
你到路由器里看看，有没有设备名叫armbian的上线了，ssh访问

liaoya, 2023-10-18:
我找到可能的原因 <https://forum.armbian.com/topic/15964-firfly-rk3399-have-not-been-booting-up-reasent-images-from-armbian/page/3/>，我用官方的 `rk3399-firefly.dtb` 替换了您提供镜像里的相应文件，两个镜像都启动成功了。

liaoya, 2023-10-18:
请教一个问题，相面的关于 emmc 的日志是什么原因，我的 firefly rk3399 的内置 emmc 不能够识别了，是硬件损坏吗？还能够恢复吗？

```
emmc reinit
mmc: ERROR: SDHCI:Transfer data timeout
mmc: ERROR: SDHCI ERR:cmd:0x153a,stat:0x0
mmc: ERROR: Tuning procedure failed, falling back to fixed sampling clock
mmc: ERROR: tuning execution failed
```

```
[    5.037957] mmc1: CQHCI version 5.10
[    5.070422] mmc1: SDHCI controller on fe330000.mmc [fe330000.mmc] using ADMA
[    5.221124] mmc1: error -110 whilst initialising MMC card
[    5.396238] mmc1: error -110 whilst initialising MMC card
[    5.575291] mmc0: Skipping voltage switch
[    5.586077] mmc1: error -110 whilst initialising MMC card
```

ophub, 2023-10-19:
我查看了下f大的6.1.y内核仓库，包含了这个文件： https://github.com/unifreq/linux-6.1.y/blob/main/arch/arm64/boot/dts/rockchip/rk3399-firefly.dts

所以打包的时候，会被覆盖了，刚才我修改了默认dtb的名字为：rk3399-firefly-150.dtb 这样防止被覆盖了。我在armbian网站看到了150提交的6.0的dts修复补丁，没找到6.1的，你看到这个firefly的6.1.y内核的补丁发给我下，更新了f大的内核仓库，这样更方便使用了。

下面是改名后重新打包的，你测试下，反馈下目前2个镜像的情况。

Armbian：
https://github.com/ophub/amlogic-s9xxx-armbian/releases/download/Armbian_jammy_save_2023.10/Armbian_23.11.0_rockchip_firefly_jammy_6.1.58_server_2023.10.18.img.gz

OpenWrt：
https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_official_save_2023.10/openwrt_official_rockchip_firefly_k6.1.58_2023.10.18.img.gz

liaoya, 2023-10-19:
> 我查看了下f大的6.1.y内核仓库，包含了这个文件： https://github.com/unifreq/linux-6.1.y/blob/main/arch/arm64/boot/dts/rockchip/rk3399-firefly.dts
> 
> 所以打包的时候，会被覆盖了，刚才我修改了默认dtb的名字为：rk3399-firefly-150.dtb 这样防止被覆盖了。我在armbian网站看到了150提交的6.0的dts修复补丁，没找到6.1的，你看到这个firefly的6.1.y内核的补丁发给我下，更新了f大的内核仓库，这样更方便使用了。
> 
> 下面是改名后重新打包的，另外u-boot也更新成了6.6里使用的版本，你测试下，反馈下目前2个镜像的情况。
> 
> Armbian： https://github.com/ophub/amlogic-s9xxx-armbian/releases/download/Armbian_jammy_save_2023.10/Armbian_23.11.0_rockchip_firefly_jammy_6.1.58_server_2023.10.18.img.gz
> 
> OpenWrt： https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_official_save_2023.10/openwrt_official_rockchip_firefly_k6.1.58_2023.10.18.img.gz
> 
> 刚才我把6.0的补丁和6.1内核比较了下，看起来能使用，我在6.1上使用了这个补丁，顺利应用和编译了。你测试完上面的镜像后，下载这个dtb文件，解压缩，覆盖`/boot/dtb/rockchip`下的同名文件，重启测试下，如果可以的话我把补丁添加到f大的仓库里。
> 
> [rk3399-firefly-150.dtb.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/13035700/rk3399-firefly-150.dtb.zip)

两个镜像都没有问题，进行了简单的测试。编译的 dtb 也可以用，有个缺陷，再串口上会不停的报
```
[   74.339445] rk3x-i2c ff160000.i2c: timeout, ipd: 0x00, state: 1
[   75.359513] rk3x-i2c ff160000.i2c: timeout, ipd: 0x00, state: 1
[   76.383514] rk3x-i2c ff160000.i2c: timeout, ipd: 0x00, state: 1
[   77.407513] rk3x-i2c ff160000.i2c: timeout, ipd: 0x00, state: 1
[   78.431514] rk3x-i2c ff160000.i2c: timeout, ipd: 0x00, state: 1
[   79.455512] rk3x-i2c ff160000.i2c: timeout, ipd: 0x00, state: 1
```
但是目前还没有发现影响功能的地方。

ophub, 2023-10-19:
好的，你确定我单独提供的 [rk3399-firefly-150.dtb.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/13035700/rk3399-firefly-150.dtb.zip) 这个测试了可以使用吧。那我给f大的仓库提交下更新。

liaoya, 2023-10-20:
> 好的，你确定我单独提供的 [rk3399-firefly-150.dtb.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/13035700/rk3399-firefly-150.dtb.zip) 这个测试了可以使用吧。那我给f大的仓库提交下更新。
> 
> 报错这是新编译的这个报错？还是之前的也报错？

是新编译的 dtb 报错。

ophub, 2023-10-21:
[rk3399-firefly.dtb.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/13059552/rk3399-firefly.dtb.zip)

你替换这个试试，下载，解压缩出来，根据你现在使用的名称修改，看看是不是有-150.

liaoya, 2023-10-24:
> [rk3399-firefly.dtb.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/13059552/rk3399-firefly.dtb.zip)
> 
> 你替换这个试试，下载，解压缩出来，根据你现在使用的名称修改，看看是不是有-150. 注意修改2个地方：`/boot/armbianEnv.txt` 和 `/boot/extlinux/extlinux.conf` 这2个文件里的dtb名字

这个 dtb 还是会不停的报，这个问题还是挺痛苦的，会导致 serial 无法很好的使用。

```
[   35.487851] rk3x-i2c ff160000.i2c: timeout, ipd: 0x00, state: 1
[   36.511916] rk3x-i2c ff160000.i2c: timeout, ipd: 0x00, state: 1
[   37.535914] rk3x-i2c ff160000.i2c: timeout, ipd: 0x00, state: 1
[   38.559849] rk3x-i2c ff160000.i2c: timeout, ipd: 0x00, state: 1
[   39.583835] rk3x-i2c ff160000.i2c: timeout, ipd: 0x00, state: 1
```

liaoya, 2023-10-24:
[amlogic-s9xxx-openwrt](https://github.com/ophub/amlogic-s9xxx-openwrt) 很好，我利用 immortalwrt 的 image builder 做出了 `rootfs`，然后用这个代码打包了自己的镜像，下次把自己的配置打包到 `rootfs` 就完美了。

closed 2023-11-05 by ophub

hkbase, 2023-11-09:
我手里也有这台机器，建议使用官方的型号名称：firefly-rk3399

ophub, 2023-11-10:
说的对，已经更正名称

liaoya, 2023-11-20:
The commit `ad6dcc5452706924c66012ca0bb582646d078465` which change name from `firefly` to `firefly-rk3399` made the image can not boot. I make sure the previous commit can build the image which can boot, and the following can't.

The UART output are 

```
SdmmcInit=2 1
mmc0:cmd5,20
SdmmcInit=0 0
BootCapSize=0
UserCapSize=29862MB
FwPartOffset=0 , 0
UsbBoot ...847574
powerOn 859850
Usb re Boot. 6859843
Usb re Boot. 12859849
Usb re Boot. 18859851
SoftReset
DDR Version 1.26 20210628
In
soft reset
SRX
Channel 0: DDR3, 800MHz
Bus Width=32 Col=10 Bank=8 Row=15 CS=1 Die Bus-Width=16 Size=1024MB
Channel 1: DDR3, 800MHz
Bus Width=32 Col=10 Bank=8 Row=15 CS=1 Die Bus-Width=16 Size=1024MB
256B stride
ch 0 ddrconfig = 0x101, ddrsize = 0x20
ch 1 ddrconfig = 0x101, ddrsize = 0x20
pmugrf_os_reg[2] = 0x32817281, stride = 0x9
OUT
Boot1 Release Time: May 29 2020 17:36:36, version: 1.26
CPUId = 0x0
```

ophub, 2023-11-20:
https://github.com/ophub/amlogic-s9xxx-armbian/commit/a0ca6ba8e4bbbcb7aa6bbd7859d4822594af7759

https://github.com/ophub/u-boot/commit/0717448f8aebdc3e9df974c60eb105bbbf638ddd

Forgot to modify u-boot and bootfs, it has been fixed

liaoya, 2023-11-20:
刷到 tf 卡上测试了，可以启动，一切正常了。非常感谢。

wackejohn, 2023-11-27:
目前，自行hack了openwrt官方snapshot，终于成功适配了firefly rk3399（没有 rk3x-i2c ff160000.i2c: timeout, ipd: 0x00, state: 1的问题，但wifi似乎ac频段无法使用），并参考其他机型的mmc sysupgrade，成功实现了保持分区表sysupgrade（目前官方rk3399的sysupgrade，每次升级，uboot、分区表都会被覆盖，并且更改分区表会导致sysupgrade失效）。。。目前正在考虑hack官方的sysupgrade，以增加uboot更新选项。。。 
附 sysupgrade过程及boot log：
```
- watchdog -
Watchdog does not have CARDRESET support
Mon Nov 27 02:40:07 CST 2023 upgrade: Sending TERM to remaining processes ...
Mon Nov 27 02:40:07 CST 2023 upgrade: Sending signal TERM to hostapd (1196)
Mon Nov 27 02:40:07 CST 2023 upgrade: Sending signal TERM to hostapd (1242)
[43829.644611] ieee80211 phy0: brcmf_cfg80211_stop_ap: SET SSID error (-110)
[43829.645254] ieee80211 phy0: brcmf_cfg80211_stop_ap: BRCMF_C_DOWN error -110
[43829.645884] ieee80211 phy0: brcmf_cfg80211_stop_ap: setting AP mode failed -110
[43829.646548] ieee80211 phy0: brcmf_cfg80211_stop_ap: BRCMF_C_UP error -110
[43829.647176] ieee80211 phy0: brcmf_set_mpc: fail to set mpc
[43829.648604] br-lan: port 2(phy0-ap0) entered disabled state
[43830.606485] br-lan: port 2(phy0-ap0) entered disabled state
Mon Nov 27 02:40:11 CST 2023 upgrade: Sending KILL to remaining processes ...
[43839.704449] stage2 (5487): drop_caches: 3
Mon Nov 27 02:40:17 CST 2023 upgrade: Switching to ramdisk...
[43841.608477] EXT4-fs (mmcblk1p1): unmounting filesystem.
[43841.610253] EXT4-fs (mmcblk1p3): unmounting filesystem.
Sun Nov 26 18:40:19 UTC 2023 upgrade: Performing system upgrade...
flashing kernel to /dev/mmcblk1p1
flashing rootfs to /dev/mmcblk1p2
[43843.433990] loop0: detected capacity change from 0 to 2040960
Format new rootfs_data at position 28770304.
[43844.242830] F2FS-fs (loop0): Found nat_bits in checkpoint
[43844.252420] F2FS-fs (loop0): Mounted with checkpoint version = 7b61248b
Saving config to rootfs_data at position 28770304.
umount: can't unmount /dev: Resource busy
umount: can't unmount /tmp: Resource busy
[43844.558887] xhci-hcd xhci-hcd.2.auto: remove, state 4
[43844.559384] usb usb4: USB disconnect, device number 1
[43844.560772] xhci-hcd xhci-hcd.2.auto: USB bus 4 deregistered
[43844.561316] xhci-hcd xhci-hcd.2.auto: remove, state 4
[43844.561782] usb usb3: USB disconnect, device number 1
[43844.563521] xhci-hcd xhci-hcd.2.auto: USB bus 3 deregistered
[43844.565170] xhci-hcd xhci-hcd.1.auto: remove, state 4
[43844.565643] usb usb2: USB disconnect, device number 1
[43844.566958] xhci-hcd xhci-hcd.1.auto: USB bus 2 deregistered
[43844.567496] xhci-hcd xhci-hcd.1.auto: remove, state 4
[43844.567962] usb usb1: USB disconnect, device number 1
[43844.569666] xhci-hcd xhci-hcd.1.auto: USB bus 1 deregistered
[43844.571201] reboot: Restarting system

U-Boot TPL 2023.10-OpenWrt-r24448-d44c037a59 (Nov 21 2023 - 23:10:31)
Channel 0: DDR3, 800MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
Channel 1: DDR3, 800MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
256B stride
Trying to boot from BOOTROM
Returning to boot ROM...

U-Boot SPL 2023.10-OpenWrt-r24448-d44c037a59 (Nov 21 2023 - 23:10:31 +0000)
Trying to boot from MMC2
spl_load_fit_image: Skip load 'atf-5': image size is 0!

U-Boot 2023.10-OpenWrt-r24448-d44c037a59 (Nov 21 2023 - 23:10:31 +0000)

SoC: Rockchip rk3399
Reset cause: RST
Model: Firefly-RK3399 Board
DRAM:  4 GiB (effective 3.9 GiB)
PMIC:  RK808 
Core:  281 devices, 28 uclasses, devicetree: separate
MMC:   mmc@fe310000: 3, mmc@fe320000: 1, mmc@fe330000: 0
Loading Environment from MMC... OK
In:    serial@ff1a0000
Out:   serial@ff1a0000
Err:   serial@ff1a0000
[... 448 lines cut ...]
[   12.378707] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4356-sdio for chip BCM4356/2
[   12.380050] usbcore: registered new interface driver brcmfmac
[   12.396858] kmodloader: done loading kernel modules from /etc/modules.d/*
[   12.573823] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4356/2 wl0: Mar  4 2020 22:06:08 version 7.35.349.87 (r724447 CY) FWID 01-c68a2890
[   17.145584] rk_gmac-dwmac fe300000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   17.148070] rk_gmac-dwmac fe300000.ethernet eth0: PHY [stmmac-0:00] driver [RTL8211E Gigabit Ethernet] (irq=POLL)
[   17.149005] rk_gmac-dwmac fe300000.ethernet eth0: No Safety Features support found
[   17.149683] rk_gmac-dwmac fe300000.ethernet eth0: PTP not supported by HW
[   17.150287] rk_gmac-dwmac fe300000.ethernet eth0: configuring for phy/rgmii link mode
[   17.152199] br-lan: port 1(eth0) entered blocking state
[   17.152676] br-lan: port 1(eth0) entered disabled state
[   17.153482] device eth0 entered promiscuous mode
[   18.656855] brcmfmac mmc0:0001:1 phy0-ap0: renamed from wlan0
[   18.755870] br-lan: port 2(phy0-ap0) entered blocking state
[   18.756367] br-lan: port 2(phy0-ap0) entered disabled state
[   18.756981] device phy0-ap0 entered promiscuous mode
[   20.459423] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
[   20.460269] br-lan: port 2(phy0-ap0) entered blocking state
[   20.460783] br-lan: port 2(phy0-ap0) entered forwarding state
[   20.461885] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   21.276416] rk_gmac-dwmac fe300000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   21.277262] br-lan: port 1(eth0) entered blocking state
[   21.277744] br-lan: port 1(eth0) entered forwarding state
```
感兴趣的朋友，可以加Q：120634404交流。。。。

alegoho, 2024-07-05:
> 刷到 tf 卡上测试了，可以启动，一切正常了。非常感谢。

你好，我刚好有这板，但我换了两个瑞芯微的刷机工具都刷不了机，请问你用那个版本的刷机工具？还有LOADER是那个？

liaoya, 2024-09-14:
> > 刷到 tf 卡上测试了，可以启动，一切正常了。非常感谢。
> 
> 你好，我刚好有这板，但我换了两个瑞芯微的刷机工具都刷不了机，请问你用那个版本的刷机工具？还有LOADER是那个？

我没有刷到 EMMC 上，就是在 tf 卡上跑。我还搞坏了一个 EMMC。

alegoho, 2024-09-15:
> > > 刷到 tf 卡上测试了，可以启动，一切正常了。非常感谢。
> > 
> > 
> > 你好，我刚好有这板，但我换了两个瑞芯微的刷机工具都刷不了机，请问你用那个版本的刷机工具？还有LOADER是那个？
> 
> 我没有刷到 EMMC 上，就是在 tf 卡上跑。我还搞坏了一个 EMMC。

我最后把系统DD进EMMC上

attintas, 2025-01-03:
问下 怎么用数据线刷openwrt啊

haha44444, 2025-01-07:
> 问下 怎么用数据线刷emmc上啊

这个只能写进tf卡刷，你可以理解为安卓刷机的线刷包和卡刷包，他俩不通用的

attintas, 2025-01-08:
> > 问下 怎么用数据线刷emmc上啊
> 
> releases中下载的文件只能写进tf卡刷，你可以理解为安卓刷机的线刷包和卡刷包，他俩不通用的 或者你可以自己试试改一改，下面的图是firefly-rk3399官网的ubuntu镜像使用rkdevtools工具解包后Output\Android\Image文件夹中rootfs.img的内容 ![image](https://private-user-images.githubusercontent.com/60396495/400949648-eca84851-3d4c-461d-b231-14d505ade335.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYzNTAyNTUsIm5iZiI6MTczNjM0OTk1NSwicGF0aCI6Ii82MDM5NjQ5NS80MDA5NDk2NDgtZWNhODQ4NTEtM2Q0Yy00NjFkLWIyMzEtMTRkNTA1YWRlMzM1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA4VDE1MjU1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdlMTRhOTRhNGFiODYxYzZhMjVhMzE1NzAyMDYzMTI0NTY4ZmJjOTNlMTFmOGE4ZTlhYmNiMzE5MzNhYTU4YTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.sm-Hd8afLUYT1Izfyh4sowP6RoEatanxV6LESbHMzIw) 下面这个图是从ophub/amlogic-s9xxx-armbian项目的releases下载的img文件中的1.primary.img文件 ![image](https://private-user-images.githubusercontent.com/60396495/400949912-93e1d274-ba4f-4610-8f08-c145bd0ec2f3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYzNTAyNTUsIm5iZiI6MTczNjM0OTk1NSwicGF0aCI6Ii82MDM5NjQ5NS80MDA5NDk5MTItOTNlMWQyNzQtYmE0Zi00NjEwLThmMDgtYzE0NWJkMGVjMmYzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA4VDE1MjU1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIxNTE3ODYzMGQ0MWIzZWZkNWFhODI3YjYwNGRjNzNlMWIwY2RjYzMyOGQ0MjBlZjhjNDljMWNlZWFjMTY3OWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.pqK3gAzvKhZzkgop1BipTISsJBQX4MeUFc5fels7q80)

哦  我看firefly的wiki上写跨固件需要用对应的刷写瑞芯微工具先清除flash才能用要刷入固件的瑞芯微工具刷固件 
那我按照上面的方法刷入sd卡在dd进emmc的话 是不是之后就不能刷回firefly的官方ubuntu固件或者android固件了？ 
如果可以的话要有什么操作步骤？

ophub, 2026-02-02:
dd写入

ophub, 2026-02-09:
我看了下armbian仓库有针对的补丁，我把补丁应用到f大的内核仓库了，这样以后编译的就是和armbian一样的了。

https://github.com/unifreq/linux-6.18.y/pull/5

6.6.y， 6.12.y， 6.18.y 这3个内核我都应用了armbian的补丁。

billy-he, 2026-09-13:
> > 感谢反馈，已经更新。
> > 我看了下armbian仓库有针对的补丁，我把补丁应用到f大的内核仓库了，这样以后编译的就是和armbian一样的了。
> > [unifreq/linux-6.18.y#5](https://github.com/unifreq/linux-6.18.y/pull/5)
> > 6.6.y， 6.12.y， 6.18.y 这3个内核我都应用了armbian的补丁。
> 
> 最新的[fnnas_rockchip_firefly-rk3399_k6.12.41_2026.02.12.img.gz](https://github.com/ophub/fnnas/releases/download/fnnas_rockchip_641/fnnas_rockchip_firefly-rk3399_k6.12.41_2026.02.12.img.gz) 641这个版本，在firefly3399上面，还是不可以，只有替换了armbian官方包的rk3399-firefly.dtb文件才能正常使用，辛苦大佬处理一下。

确实，我也板子也是这情况，用最新openwrt_immortalwrt_rockchip_firefly-rk3399_k6.18.48_2026.09.01.img.gz镜像也启动不了，麻烦老大看看。

SPWangFei, 2026-09-13:
您好，您发送给我的邮件已经收到，我将尽快处理~

ophub, 2026-09-16:
我修复了一下6.18的内核，你下载最新的试试