ophub/amlogic-s9xxx-openwrt #499: 最新的docker版OP，webui好像进不去

iJayscn, 2023-11-04:
**Device Information | 设备信息**
 - SOC: [e.g. S095x3]S905D
 - Model [e.g. HK1] N1

**OpenWrt Version | 系统版本**
 - Kernel Version: Linux 6.1.38-ophub
 - Release:DOCKER

SSH 正常，ping OP容器正常，容器内ping 其他网站都正常

就是webui进不去，不知道啥问题

iJayscn, 2023-11-07:
DIGEST
eba43034216e

这个版本也不行，是只有我有这个问题吗？ 大家都正常吗？

iJayscn, 2023-11-17:
> 修改 /etc/config/uhttpd 文件，注释掉 2 行 listen_https

这样是能解决

后续版本会修复不？

closed 2023-11-17 by iJayscn

ophub, 2023-12-01:
切换到官方源的tar包使用吧，lede

[image]
[image]
[image]