ophub/amlogic-s9xxx-openwrt #501: M301A S905l2盒子的问题

iveryo, 2023-11-08:
**Device Information | 设备信息**
 - SOC: [e.g. S095l2]
 - Model [e.g. M301A]

**OpenWrt Version | 系统版本**
 - Kernel Version: [e.g. 5.15.100]
 - Release: [e.g. official/lede/immortalwrt]

**Describe the bug | 问题描述**
问题一
M301A S905L2盒子，能安装的最高版本是5.15.137，通过U盘可以启动。
但用6.1.xx版本通过U盘启动时黑屏，并不断重启。

问题二
5.15.137版本通过U盘启动后安装OpenWrt，选择 【[ 116 ] MGV2000,MGV3000,M301A】安装后启动黑屏。
最终测试选择【[ 124 ] M302A,M304A,CM211-1,CM311-1】安装后可启动。
目前状态页显示的型号为 Amlogic Meson GXL (S905L3b) MBH-M302A Box :)

问题三
1.拿immortalwrt版本的固件来说，在网络存储的【挂载 SMB/CIFS 网络共享文件夹】，这个设置挂载不成功。
看日志 提示 CIFS: VFS: No username specified
但在提交时用户名是存在的，你可重现一下。
目前我用命令进行挂载是可以的，如果这个软件有效就更好管理了。

2.以及网络存储的【网络共享】，使用时也没有效果。这个没有看日志，我用的服务下的【网络共享】可以解决。
如果没有用就删掉吧。

以上发了不少时间测试，为什么我执意要升到6.1以上呢。在安装一些软件包时一直提示内核需要6.1以上，
老多软件包安装不上，科学上网也没有安装上，不知大神有何建议。

iveryo, 2023-11-10:
**3和4的问题基本解决，解决的方法**
1. 使用SSH连接或WEB中的终端进行安装
2. 在安装软件时不要一股的全部拷贝多行指令执行安装依赖，某些库升级后反而出更多问题（我测试过多次，破坏原有环境）
3. 可在没有进行opkg update前执行一次安装依赖，看具体系统中缺少哪些包，保留系统存在库的原有版本，只针对这几个没有的库进行安装

这是最大机率安装成功的方法（我觉得），大神你怎么看！
问题关闭吧，问题1和2并不影响使用了，大神知会就行。

closed 2023-11-10 by iveryo

mrsoshining, 2025-01-18:
> 3\. 没有的库进行安装
大佬，怎么针对没有的库进行安装

iveryo, 2025-01-18:
> > 3. 没有的库进行安装
> > 大佬，怎么针对没有的库进行安装

特指定下面这几个依赖的库进行安装能解决
opkg install libcap-bin
opkg install ruby
opkg install ruby-yaml
opkg install kmod-inet-diag
opkg install libmbedtls

mrsoshining, 2025-01-18:
> > > 3. 没有的库进行安装
> > >    大佬，怎么针对没有的库进行安装
> 
> 特指定下面这几个依赖的库进行安装能解决 opkg install libcap-bin opkg install ruby opkg install ruby-yaml opkg install kmod-inet-diag opkg install libmbedtls

无论我是用officle还是imm wrt,都无法安装openclash