ophub/amlogic-s9xxx-openwrt #508: 荐片盒子ssh用不了！？

y9858, 2023-11-19:
**Device Information | 设备信息**
 - SOC: [入库566]
 - Model [荐片TV]

**OpenWrt Version | 系统版本**
 - Kernel Version: [6.1.62]
 - Release: [immortalwrt]

**Describe the bug | 问题描述**
使用immortalwrt master源码编译，用本源码打包。网页ttyd和电脑MobaXterm连ssh都提示如图：

[image]

按键盘ctrl+c/enter也没反应！。

ophub, 2023-11-19:
你编译的底包缺必要依赖，这个欢迎界面的温度等都没显示出来：
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/make-openwrt/openwrt-files/common-files/etc/profile.d/30-sysinfo.sh

https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/documents/README.cn.md
看10.11

y9858, 2023-11-19:
[kmod-brcmfmac(SDIO)|kmod-brcmutil|kmod-cfg80211|kmod-mac80211|hostapd-common|wpa-cli|wpad-basic|iw|zoneinfo-europe (other) other找不到 没选 |acpid]用不上wifi 这些都没选。情况依旧。附上.config。
[1.txt](https://github.com/ophub/amlogic-s9xxx-openwrt/files/13402530/1.txt)

y9858, 2023-11-20:
换大雕源码 ok了

closed 2023-11-20 by y9858