ophub/amlogic-s9xxx-openwrt #518: Rootfs partition does not exit in emmc

SHINRA-OPI, 2023-12-17:
**Device Information | 设备信息**
 - SOC: [s905x]
 - Model [Generic Tv Box(tv98) 1gb/8gb emmc]

**OpenWrt Version | 系统版本**
 - Kernel Version: [6.1.65]
 - Release: [lede]

There are no problem if the openwrt was booted in usb, the problem will occur after the installation in emmc. In booting, there is an alert "UUID=(the long id) does not exist. Dropping to shell! " Then it stuck in "(initramfs)" . I enter blkid command and there is no rootfs partition only "EMMC SHARE, EMMC ROOTFS1, EMMC ROOTFS2 AND EMMC BOOT". I try to plugged in my usb drive and enter exit command the booting continues like normal. I will provide screenshot if you need tomorrow.

SHINRA-OPI, 2023-12-18:
[image]

SHINRA-OPI, 2023-12-18:
[image]

ophub, 2023-12-18:
You cannot see the picture you posted. When posting an image, you need to wait for the image to finish uploading before submitting it. Without the UUID, it indicates that the write was not successful. Please try writing to the eMMC again (System Menu - Amlogic Service - Install OpenWrt).

SHINRA-OPI, 2023-12-18:
[image]
[image]

ophub, 2023-12-18:
System Menu - Amlogic Service - Install OpenWrt

SHINRA-OPI, 2023-12-18:
I try it several times it still the same error

ophub, 2023-12-19:
try kernel 5.15.y

SHINRA-OPI, 2023-12-27:
the 2023.12,23 version of lede with kernel 5.15.144 is now booting without a problem. Thank you

closed 2024-01-07 by ophub