ophub/amlogic-s9xxx-openwrt #521: immortalwrt 防火墙区域没有 开启fullcone选项

Link-ee, 2023-12-26:
如题，使用23.05和master  fw4 打包出来的在盒子上没有开去fullcone选项，下图是我X86使用的时候可以正常显示。
[image]

ophub, 2023-12-26:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/ab863ef755efeca2f5fe4a5d461fff926bfecd3e/config/immortalwrt-master/config#L2386

把2386行这个改成y试试

`CONFIG_PACKAGE_kmod-ipt-fullconenat=y`

Link-ee, 2023-12-26:
> https://github.com/ophub/amlogic-s9xxx-openwrt/blob/ab863ef755efeca2f5fe4a5d461fff926bfecd3e/config/immortalwrt-master/config#L2386
> 
> 把2386行这个改成y试试
> 
> `CONFIG_PACKAGE_kmod-ipt-fullconenat=y`

这个是ipt的吧，默认的nft的fullcone是已经选中了
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/ab863ef755efeca2f5fe4a5d461fff926bfecd3e/config/immortalwrt-master/config#L2454

closed 2024-01-07 by ophub

AoXIUy, 2024-06-28:
老哥，无法开启Fullcone解决了吗