ophub/amlogic-s9xxx-openwrt #529: ZTE B860H

adereksisusanto, 2024-01-10:
**Device Information | 设备信息**
 - SOC: Amlogic s905x
 - Model ZTE B860H

**OpenWrt Version | 系统版本**
 - Kernel Version: 5.15.145
 - Release: official

**Describe the bug | 问题描述**
- [x] for zte b860h firmware openwrt_official_amlogic_s905x-t95_k5.15.145_2024.01.07.img.gz wireless AP is activated can't login to LUCI and wireless remains undetected

- [x] for zte b860h firmware openwrt_immortalwrt_amlogic_s905x_k5.15.145_2024.01.07.img can't login to LUCI and LAN is not connected

Please provide a detailed description of the issue and, if necessary, attach a screenshot.
详细描述问题，并在必要时附上屏幕截图。

ophub, 2024-01-11:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/1079997d5e79596c54bcf31ebd508969f813fa59/make-openwrt/openwrt-files/common-files/etc/model_database.conf#L47

Compile the firmware by selecting the correct board option `s905x-b860h`

adereksisusanto, 2024-01-13:
terminal not access
[image]

ophub, 2024-01-14:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/c5c44fe5df80de8d58be773c08326275d13f414c/make-openwrt/openwrt-files/common-files/etc/profile.d/30-sysinfo.sh#L18-L23

adereksisusanto, 2024-01-14:
how to fix  @ophub ?

ophub, 2024-01-14:
try lede tag

https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_lede_save_2024.01

adereksisusanto, 2024-01-14:
thanks @ophub 
[image]

closed 2024-01-14 by adereksisusanto