ophub/amlogic-s9xxx-openwrt #531: T95 Max Support?

aptonline, 2024-01-13:
**Device Information | 设备信息**
 - SOC: Allwinner
 - Model: H6

**OpenWrt Version | 系统版本**
 - Kernel Version: 5.15.y
 - Release: openwrt_official_allwinner_tanix-tx6_k5.15.145_2024.01.07

**Describe the bug | 问题描述**
Tried to boot the T95 Max device using the about firmware and can successfully reach the openwrt screen but verbose complains about file system may be corrupt and advices to run fcsk.

no login screen shown and no access point created.

any ideas how to get this working?

[image]

ophub, 2024-01-14:
try lede tags

aptonline, 2024-01-14:
Ok that’s better, no complaints of corrupted storage and get the OpenWRT prompt (don’t have to login just press ‘enter’).

still no ‘openwrt’ access point created and not showing up on network if I use LAN cable.

I can enter ‘help’ at the command prompt and see and interact ok.

[image]

aptonline, 2024-01-18:
Any idea how to get the access point working?

ophub, 2024-01-18:
Please search online for the DTS file of this device.

aptonline, 2024-01-18:
> Please search online for the DTS file of this device.

Hi, I found this online with appears to include the .dts file.

Should I try the lede tag build with this .dts file?

aptonline, 2024-01-18:
Tried the .dts file in the previously provided lede build and same issue.

noticed you have now removed the lede builds?

aptonline, 2024-01-21:
Hi, What should I do with the .dts file I found?

closed 2024-01-27 by ophub

aptonline, 2024-01-27:
Hi, I see you’ve closed the issues but still not working for me. What should I do with the .dts file?

yusrang, 2024-07-19:
> Please search online for the DTS file of this device.

Hi, I have the same issue, as its ethernet port is not working. Current driver is rtl8822cs-bt.

Here attached, dts file for Tanix TX6.
https://drive.google.com/file/d/1hHpF-7JFm2GyJAwYzI7O1cCVcRaPEX2Q/view?usp=sharing

Sincerely