ophub/amlogic-s9xxx-openwrt #533: 2024.01 release的openwrt_official_amlogic_s905x3无法使用

kousyougi, 2024-01-16:
**Device Information | 设备信息**
 - SOC: s905x3
 - Model x96air

**OpenWrt Version | 系统版本**
 - Kernel Version: k6.1.70
 - Release: official

**Describe the bug | 问题描述**
自己编译的无法使用。使用release发布的也出现一样的错误信息。
https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_official_save_2024.01
openwrt_official_amlogic_s905x3_k6.1.70_2024.01.07.img.gz
[image]

tramplong, 2024-01-17:
TVI3315a同样的问题

kousyougi, 2024-01-17:
最近多個issue可能都是一樣的問題

https://github.com/ophub/amlogic-s9xxx-openwrt/issues/530

https://github.com/ophub/amlogic-s9xxx-openwrt/issues/511

https://github.com/ophub/amlogic-s9xxx-openwrt/issues/510

https://github.com/ophub/amlogic-s9xxx-openwrt/issues/508

https://github.com/ophub/amlogic-s9xxx-openwrt/issues/534

allentd, 2024-01-17:
Looks like all 3 latest opened issues is about this, hope it can get fixed.

Same for me on Minix U1 (s905).

ophub, 2024-01-18:
可能官方的源码有点问题，我删除之前的official这个tags重新编译了一下，再试试

allentd, 2024-01-18:
> 可能官方的源码有点问题，我删除之前的official这个tags重新编译了一下，再试试

Thanks! Confirmed it's fixed.

Thank you for your hard work all this time and also for this fix.

kousyougi, 2024-01-20:
安裝了可以用
但 執行 openwrt-update-amlogic 會跳出一樣的訊息。

root@OpenWrt:~# openwrt-update-amlogic
Current device: X96 MAX PLUS Q2, X96 Air Q1000 [ amlogic ]
Cannot find the partition corresponding to the root file system!
root@OpenWrt:~#

closed 2024-01-27 by ophub

kousyougi, 2024-01-29:
參考前面的patch，修改 /usr/sbin/openwrt-update-amlogic 可以用了。

[overlay.patch](https://github.com/ophub/amlogic-s9xxx-openwrt/files/14084536/overlay.patch)