ophub/amlogic-s9xxx-openwrt #535: M401A写入emmc时设备要选cm311-1a-yst, 并且ssh无法连接

weihu120, 2024-01-17:
**Device Information | 设备信息**
 - SOC: [s905l3a]
 - Model [M401A]

**OpenWrt Version | 系统版本**
 - Kernel Version: [5.15.145-ophub]
 - Release: [OpenWrt SNAPSHOT r0-9658050 / LuCI Master git-24.006.68576-8debeeb]

**Describe the bug | 问题描述**。
写入U盘可以启动，但ssh无法连接。
[image]
另外，写入emmc时选m401a无法启动，接上显示器，开机屏幕闪几下然后一直灰屏，整过过程看不到任何输出，后来先cm311-1a-yst到是可以启动，但写入后ssh还是无法连接，web上的终端也无法连接。
[image]

ophub, 2024-01-18:
我重新编译了一下官方版，你再试试

weihu120, 2024-01-18:
> 我重新编译了一下官方版，你再试试

刚试了晚上的版本，ssh是没有问题了，但是写入emmc时选m401a还是挂了，写入后不能启动。

weihu120, 2024-01-18:
写入emmc时选m401a和cm311-1a-yst两者之间执行的操作有些什么区别？

ophub, 2024-01-19:
/boot/uEnv.txt使用的dtb不同，使用的u-boot不同

closed 2024-01-27 by ophub