ophub/amlogic-s9xxx-openwrt #537: 拉取tag编译时会被重置架构，拉取banches则不会。

1877574385, 2024-01-27:
Device Information | 设备信息

SOC: [e.g. S095x3]
Model [x96max +]
Armbian Version | 系统版本

Kernel Version: [e.g. 5.15.100]
Release: [e.g. jammy/bullseye]
Describe the bug | 问题描述
拉取tag编译时会被重置架构，拉取banches则不会。
默认状态时会被重置，
然后我把yaml改成如下
# Clone source code
git clone -q --single-branch --depth=1 --branch=${REPO_BRANCH} ${REPO_URL} openwrt &&
cd openwrt &&
git reset --hard &&
git checkout -b main1
ln -sf /workdir/openwrt ${GITHUB_WORKSPACE}/openwrt

同样会被重置。armsr被重置到ath79

ophub, 2024-01-27:
哪个？

1877574385, 2024-01-27:
> 哪个？

x96max+ 用openwrt 官方源码云编译的时候  ，如果REPO_BRANCH="v23.05.0" 这种类型（tags）时，编译成功后的架构被重置到初始状态ath79.

1877574385, 2024-01-27:
config 里面设置的是CONFIG_TARGET_armsr=y  ,编译出来就变成 CONFIG_TARGET_ath79=y

ophub, 2024-01-27:
使用最新的brancher:main是armsr，其他旧版的自己配置config

1877574385, 2024-01-27:
打包的包内核需要和openwrt的内核一致吗？因为默认openwrt的内核好难调成中间版本。默认都到最新

ophub, 2024-01-27:
内核随意。

closed 2024-01-27 by ophub