ophub/amlogic-s9xxx-openwrt #538: 晶晨宝盒在线更新报错： Failed to get Image name: []

0xDarker, 2024-01-28:
**Device Information | 设备信息**
 - SOC: [S095x3]
 - Model [HK1Box]

**OpenWrt Version | 系统版本**
 - Kernel Version: [6.1.62]
 - Release: [immortalwrt]

**Describe the bug | 问题描述**
晶晨宝盒在线更新下载完点击update直接报错
`
 [[1;91m Error [0m] Failed to get Image name: [ ]
`

[image]

在手动更新那里点也是
[image]

ophub, 2024-01-30:
`curl -fsSL git.io/luci-app-amlogic | bash`

在ttyd终端里升级插件到最新版再试试。

0xDarker, 2024-01-30:
> `curl -fsSL git.io/luci-app-amlogic | bash`
> 
> 在ttyd终端里升级插件到最新版再试试。你使用的是哪个源码编译的？
更新到最新3.1.220-1 了还是一样的问题。我是下载的releases里的`openwrt_immortalwrt_amlogic_s905x3_k6.1.62_2023.11.12.img.gz`，
机型是HK1Box

ophub, 2024-01-30:
[image]

我升级到最新版，把内核和固件都更新测试了，没问题啊

0xDarker, 2024-01-30:
一开始全新安装上之后我这也是能更新成功的，后来更新了几次之后就报这个错了

ophub, 2024-01-30:
[image]

[image]

我把仓库的3个不同源的固件都测试了，没问题，上面这是immortalwrt更新内核。

0xDarker, 2024-01-30:
报这个错的话是哪里不对导致的呢

ophub, 2024-01-30:
https://github.com/ophub/luci-app-amlogic/blob/ae08afbfb23a22ff63f0c7bc96aef960f285d86e/luci-app-amlogic/root/usr/sbin/openwrt-kernel#L79-L82

你ttyd执行 `ls /boot` 看看，缺少文件： `*Image`

0xDarker, 2024-01-30:
> https://github.com/ophub/luci-app-amlogic/blob/ae08afbfb23a22ff63f0c7bc96aef960f285d86e/luci-app-amlogic/root/usr/sbin/openwrt-kernel#L79-L82
> 
> 你ttyd执行 `ls /boot` 看看，缺少文件： `*Image`

要晚上回去看看了，刚才在luci上点重启没有启起来，这个问题也是经常出现需要重新插拔电源。

YuFong, 2024-01-30:
@0xDarker 可以分享一下磁盘挂载点的截图吗？我的有点搞乱了，参考一下，谢谢。

liker5092, 2024-01-30:
怎么降级呢？安装 v2raya总是报错  内核错误  需要6.1.71.xxxxxx

0xDarker, 2024-01-30:
> @0xDarker 可以分享一下磁盘挂载点的截图吗？我的有点搞乱了，参考一下，谢谢。

不同型号设备挂载点不一样你看我的没有用

0xDarker, 2024-01-30:
> https://github.com/ophub/luci-app-amlogic/blob/ae08afbfb23a22ff63f0c7bc96aef960f285d86e/luci-app-amlogic/root/usr/sbin/openwrt-kernel#L79-L82
> 
> 你ttyd执行 `ls /boot` 看看，缺少文件： `*Image`

@ophub 大佬我看了/boot下是空的

ophub, 2024-01-30:
/boot挂载点掉了？手动挂上

0xDarker, 2024-01-30:
> /boot挂载点掉了？手动挂上

谢大佬，不知为啥/boot的卷标变成了SD_BOOT，改了设备卷标之后更新内核成功了

closed 2024-01-30 by 0xDarker