ophub/amlogic-s9xxx-openwrt #539: JP-Tvbox的HCY3655无线网卡驱动怎么安装？

vpei, 2024-01-31:
**Device Information | 设备信息**
 - SOC: [e.g. S095x3]
 - Model [e.g. HK1]

**OpenWrt Version | 系统版本**
 - Kernel Version: [e.g. 5.15.100]
 - Release: [e.g. official/lede/immortalwrt]

**Describe the bug | 问题描述**
All issues will only remain open for one week to prioritize resolving them.
After that period, they will be closed but can still continue to be discussed in the thread.
所有的问题都只保留一周的开启状态供重点关注解决，超时将关闭，但仍然可以长期继续跟帖讨论。

Please provide a detailed description of the issue and, if necessary, attach a screenshot.
详细描述问题，并在必要时附上屏幕截图。

JP-Tvbox的HCY3655无线网卡驱动怎么安装？

EvenEnd, 2024-02-03:
用flippy 6.1.75内核即可

vpeu, 2024-02-03:
> 用flippy 6.1.75内核即可

这个好。👍

vpei, 2024-02-03:
> 用flippy 6.1.75内核即可

我来试试，谢谢。

vpei, 2024-02-05:
> 用flippy 6.1.75内核即可

的确可以用，用起来不错。可以直接做无线路由。

closed 2024-02-05 by vpei