ophub/amlogic-s9xxx-openwrt #541: R3300-L小红盒移动版，用当前最新的5.15.148U盘可以启动，写入emmc就后无法启动

bird202301, 2024-02-02:
尊敬的O大您好，我的问题是：
我的R3300-L小红盒移动版，用当前最新的5.15.148U盘可以启动，运行正常。
当我试图写入emmc就后无法启动，但是观察到hdmi有信号，屏幕不待机，就是黑屏。
我也把盒子实验性写入了其他玩家提供的2个img，USB设备就可以启动，但依然在写入EMMC后不能启动。
我联想到我的R3300-L在写Armbian进入EMMC的时候，最开始也是黑屏，后来加了参数 armbian-install -m yes 成功了，是不是这个原因呢？
openwrt不知道命令行带不带这个-m yes 参数呢？

谢谢您。

ophub, 2024-02-02:
`openwrt-install-amlogic yes`
或者
系统---晶晨宝盒--插件设置选择使用bootloader，然后在安装openwrt里选择你的设备型号点安装即可。
这2种结果相同都是使用主线

bird202301, 2024-02-02:
选择使用插件bootloader的方法在您以前的帖子中看到了，忘说了，无效，抱歉啊。
我用命令行试了很多次,问题依旧，不能启动。
我想了个办法，哪位有成功写入到emmc的吗？我自己用DDBR恢复到我的盒子里面，看能不能成功。再次感谢O大的回复提示

bird202301, 2024-02-02:
同一个盒子R3300-L，Armbian可以写入EMMC，openwrt写入启动不起来，感觉这个问题可以解决。

bird202301, 2024-02-03:
向O大汇报最新进展，结果是刷别代码的的成功！在您的提示下，
我自己昨晚到今天一直在实验，刷失败了不亮顶多刷入安卓再来，后来发现7JI大佬提取的精简残废救砖rom恰好拿来做实验
陆续刷了几十款代码，测试中有几个能残缺启动（125.126.128等），越刷越勇，终于让我试验出了能运行的代码，就是127代表型号我都没听过呢，刷了之后看到启动信息里面显示是E900V22E，可我这明明是移动绿色电路板的R3300L啊，所以我琢磨代码124也写着E900V22E啊，它是正宫啊，所以我就继续刷了124，结果和猜测一样，也能运行，至此问题初步解决。先试试124啦，
总结起来就是插件选择更新bootloader了，然后就是一轮轮试验啦，124 127均可，没做进一步测试。

ophub, 2024-02-03:
好的，感谢分享，能通刷挺好的👌

18271211, 2024-02-05:
> 向O大汇报最新进展，结果是刷别代码的的成功！在您的提示下， 我自己昨晚到今天一直在实验，刷失败了不亮顶多刷入安卓再来，后来发现7JI大佬提取的精简残废救砖rom恰好拿来做实验 [...]

nb，早点看到就不会多浪费两小时了。

18271211, 2024-02-05:
> nb，早点看到就不会多浪费两小时了。

按照楼上刷的124，成功，然后ssh内还是认出了3300L，原理不知
[image]

bird202301, 2024-02-05:
我刷5.15.XXX的原始目的是它能够支持无线AP，这就好玩了，但是看HDMI输出有非常频繁RTL871X的信息，这个问题看有人的帖子里也提及过，我觉得非常占用系统资源，此时我的系统占用平均也要30%左右，就这样烤机12小时，第二天早上看输出又有新的变化，错误更多了，一屏一屏输出，而且我的无线网络崩溃，已经建立的AP接入点点击连接也连接不上去，还经常性消失，此时系统占用飙升至65-70%，烤机期间我啥都没做，Transmission服务没启动，只配置了有线网卡为固定ip和无线网卡接入点的密码。不排除我机器的体质不行，和Armbian比较起来，完全没有Armbian稳定，但是Armbian没无线网卡驱动（不知说的对不对，我比较菜5.15.XXX按理应该也有啊，我刷的6.1.69也能玩AP），不能玩AP，armbian下的tr也工作的好好的，所以近期我又转回了Armbian，把我的小红盒子变成专职下载机了。话说DDBR大法太好了

bird202301, 2024-02-05:
> nb，早点看到就不会多浪费两小时了。
期待你的测试结果啊，上面我反馈了一些测试信息

跑到你的网站看了看，第一个6.0固件我有，第二个好玩吗？

最后还是感谢O大的辛勤劳作，我们才能有这样好的系统玩。

closed 2024-02-05 by bird202301

mccsss, 2025-10-19:
同样U盘可以启动，刷emmc就不能用了，现在有解决方法了吗