ophub/amlogic-s9xxx-openwrt #543: 请问各位的Transmission运行正常吗？我的就是不开始下载

bird202301, 2024-02-08:
请问各位的Transmission运行正常吗？我的就是不开始下载

机器型号R3300L小红盒，按照型号应该选用129，但129写入黑屏，用127号代码写入EMMC运行基本正常，无线网卡驱动似乎有点问题，HDMI输出日志较多，平均一秒能达到10条以上，以上是基本情况。
我的tr添加了种子，设置了下载目录，目录是挂载的外接移动硬盘，设置了共享目录，windows访问正常，权限全部设置了root用户和777权限，但是种子文件添加后就是提示连不上tracker，黄色叹号，特意选了几个网站的热门种子，虽然我是内网，但windows上都可以浏览网站，无异常，而且openwrt的AP功能也好用，不能上网的原因，同一内网的windows下的QB可以下载，排除了IPS封锁，现在无解了，请大家帮忙想想可能的线索和办法，提前祝大家春节快乐！

bird202301, 2024-02-08:
在系统 /软件包 点击更新也是不能连接内置的源，移动，联通网络都实验了，想换个软件或者版本更新的路也走不通了

bird202301, 2024-02-09:
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/targets/armvirt/64/packages/Packages.gz
*** Failed to download the package list from https://mirrors.cloud.tencent.com/lede/snapshots/targets/armvirt/64/packages/Packages.gz

Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_generic/base/Packages.gz
*** Failed to download the package list from https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_generic/base/Packages.gz

Downloading https://mirrors.cloud.tencent.com/lede/releases/18.06.9/packages/aarch64_generic/luci/Packages.gz
*** Failed to download the package list from https://mirrors.cloud.tencent.com/lede/releases/18.06.9/packages/aarch64_generic/luci/Packages.gz

Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_generic/packages/Packages.gz
*** Failed to download the package list from https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_generic/packages/Packages.gz

Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_generic/routing/Packages.gz
*** Failed to download the package list from https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_generic/routing/Packages.gz

Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Collected errors:
 * opkg_download: Failed to download https://mirrors.cloud.tencent.com/lede/snapshots/targets/armvirt/64/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_generic/base/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://mirrors.cloud.tencent.com/lede/releases/18.06.9/packages/aarch64_generic/luci/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_generic/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_generic/routing/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 软件包列表已超过 24 小时未更新

bird202301, 2024-02-09:
我的思路是既然内置的tr不适合我，要不升级下版本，要不换QBittorent，然而软件包这里又卡住了，无奈我基础太差了，只能baidu看看前人的经验，慢慢琢磨啦，Arm变已经成功把盒子变成了PT下载机，看看OPenwrt能不能实现AP热点（成功）+下载机（卡住了）+FTP(成功）+网络共享（成功），万望这里的大侠给与指导，再次祝大家春节快乐啊

bird202301, 2024-02-11:
我用F大最新版本5.15.148 给N1编译的版本tr能用，不用特别设置，但是无线不能用！
这要是嫁接O大的无线过去就完美了

bird202301, 2024-02-15:
Downloading https://mirrors.ustc.edu.cn/lede/snapshots/targets/armvirt/64/packages/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/lede/snapshots/targets/armvirt/64/packages/Packages.gz

Downloading https://mirrors.ustc.edu.cn/lede/snapshots/packages/aarch64_generic/base/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/lede/snapshots/packages/aarch64_generic/base/Packages.gz

Downloading https://mirrors.ustc.edu.cn/lede/releases/18.06.9/packages/aarch64_generic/luci/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/lede/releases/18.06.9/packages/aarch64_generic/luci/Packages.gz

Downloading https://mirrors.ustc.edu.cn/lede/snapshots/packages/aarch64_generic/packages/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/lede/snapshots/packages/aarch64_generic/packages/Packages.gz

Downloading https://mirrors.ustc.edu.cn/lede/snapshots/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://mirrors.ustc.edu.cn/lede/snapshots/packages/aarch64_generic/telephony/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/lede/snapshots/packages/aarch64_generic/telephony/Packages.gz

Collected errors:
 * opkg_download: Failed to download https://mirrors.ustc.edu.cn/lede/snapshots/targets/armvirt/64/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://mirrors.ustc.edu.cn/lede/snapshots/packages/aarch64_generic/base/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://mirrors.ustc.edu.cn/lede/releases/18.06.9/packages/aarch64_generic/luci/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://mirrors.ustc.edu.cn/lede/snapshots/packages/aarch64_generic/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://mirrors.ustc.edu.cn/lede/snapshots/packages/aarch64_generic/telephony/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

换了源也是这样，这其中我哪里出问题了，请大家提示，谢谢。

bird202301, 2024-02-15:
我又跑回来研究O大的版本了，TR能用就ok啦，docker也拉不回来镜像啊

ophub, 2024-02-16:
docker也有国内源，你换源

或者你理解一下这个意思：🪜

closed 2024-02-29 by bird202301