ophub/amlogic-s9xxx-openwrt #547: 请问上哪才能找到合适的插件，找了一天了，没一个能安装上的

Twodogs2020, 2024-02-20:
型号 | CM311-1a-YST
架构 | ARMv8 Processor rev 4
目标平台 | armsr/armv8
固件版本 | OpenWrt SNAPSHOT r0-79888f9 / LuCI Master git-24.044.66292-9589392
内核版本 | 6.6.15-happy-new-year

ophub, 2024-02-22:
for后自己添加插件，编译

Twodogs2020, 2024-03-04:
还有，在S905L3的盒子上，用的是CM211-1，更改了接口eth0的DNS，始终PING不通外网，找了很多帖子，试了他们的方法，还是不行

Twodogs2020, 2024-03-04:
> for后自己添加插件，编译

Twodogs2020, 2024-03-05:
> for后自己添加插件，编译

手动安装了插件，但是主页上显示不出来

ophub, 2024-03-05:
immortalwrt和官方openwrt有上游维护，方面在线安装软件。lede的无法在线安装。

Twodogs2020, 2024-03-08:
> for后自己添加插件，编译
自己按照文档编了个镜像，结果刷进去一直报错：
[  128.100530] RTW: ERROR sdio_io: bSurpriseRemoved, skip write 0x00042, 1 bytes
[  128.107601] RTW: ERROR _halmac_sdio_cmd52_write: I/O FAIL!

ophub, 2024-03-08:
两个错误都指向一个共同的情况：SDIO接口上的无线设备出现了通信问题。这些错误可能是由于多种原因造成的，包括但不限于：

- 物理连接不稳定或零部件损坏，导致SDIO设备意外断开（被物理移除或丢失电源）。
- 驱动程序BUG，可能是由于错误的状态处理或者与该特定硬件版本不兼容。
- 硬件故障或不兼容问题。

解决方法，禁用无线

Twodogs2020, 2024-03-08:
> 两个错误都指向一个共同的情况：SDIO接口上的无线设备出现了通信问题。这些错误可能是由于多种原因造成的，包括但不限于：
> 
> * 物理连接不稳定或零部件损坏，导致SDIO设备意外断开（被物理移除或丢失电源）。
> * 驱动程序BUG，可能是由于错误的状态处理或者与该特定硬件版本不兼容。
> * 硬件故障或不兼容问题。
> 
> 解决方法，禁用无线

谢谢！

Twodogs2020, 2024-03-11:
> > 两个错误都指向一个共同的情况：SDIO接口上的无线设备出现了通信问题。这些错误可能是由于多种原因造成的，包括但不限于：
> > 
> > * 物理连接不稳定或零部件损坏，导致SDIO设备意外断开（被物理移除或丢失电源）。
> > * 驱动程序BUG，可能是由于错误的状态处理或者与该特定硬件版本不兼容。
> > * 硬件故障或不兼容问题。
> > 
> > 解决方法，禁用无线
> 
> 谢谢！

[image]
根据文档，修改了这个配置文件，但是编译后的包任然不显示对应的插件

ophub, 2024-03-11:
这是包，不是插件界面，你要界面的话打开对应的luci-app-xxx

Twodogs2020, 2024-03-12:
> 这是包，不是插件界面，你要界面的话打开对应的luci-app-xxx

好的，我试试，谢谢您！

Twodogs2020, 2024-03-12:
> 这是包，不是插件界面，你要界面的话打开对应的luci-app-xxx

对了，那些包是啥作用啊，打开后跟没打开啥区别呀？

shirakun, 2024-03-12:
> > 这是包，不是插件界面，你要界面的话打开对应的luci-app-xxx
> 
> 对了，那些包是啥作用啊，打开后跟没打开啥区别呀？

luci-app开头一般是跟管理页面有关系
后面带zh/en/jp之类的一般都是语言包
然后kmod是和内核相关的包
其它的是普通的包
每个包的用处应该都会在它的desc里给出
很多包都有依赖关系,依赖关系在欲安装的包的dep里面也有给出
此外也可以google搜索包名查找

(通常每个项目的issues只讨论和项目相关的问题,像这类问题推荐去openwrt官方论坛或者恩山无线论坛或是各种交流群讨论,通常这些论坛/交流群能获得更及时的回复