ophub/amlogic-s9xxx-openwrt #550: Docker -> Network 然后报错

MaxCrazy1101, 2024-03-02:
**Device Information | 设备信息**
 - SOC: [e.g. S095x3] rk3399
 - Model [e.g. HK1] fine3399

**OpenWrt Version | 系统版本**
 - Kernel Version: [e.g. 5.15.100] 5.15.149
 - Release: [e.g. official/lede/immortalwrt] immortalwrt

**Describe the bug | 问题描述**
All issues will only remain open for one week to prioritize resolving them.
After that period, they will be closed but can still continue to be discussed in the thread.
所有的问题都只保留一周的开启状态供重点关注解决，超时将关闭，但仍然可以长期继续跟帖讨论。
Docker -> Network 然后报错
Please provide a detailed description of the issue and, if necessary, attach a screenshot.
详细描述问题，并在必要时附上屏幕截图。
[image]

wuyuqq, 2024-03-07:
一样有这个问题

/usr/lib/lua/luci/dispatcher.lua:482: Failed to execute cbi dispatcher target for entry '/admin/docker/networks'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/dockerman/networks.lua:44: attempt to index field 'Config' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:482: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:129: in function </usr/lib/lua/luci/dispatcher.lua:128>

MaxCrazy1101, 2024-03-10:
> 一样有这个问题
>
> /usr/lib/lua/luci/dispatcher.lua:482: Failed to execute cbi dispatcher target for entry '/admin/docker/networks'. The called action terminated with an exception: [...]

请问这个问题你解决了吗

wuyuqq, 2024-03-10:
> 请问这个问题你解决了吗

没有解决，去lede仓库看了下 也有同样的情况，说是上游的问题。

MaxCrazy1101, 2024-03-10:
> 没有解决，去lede仓库看了下 也有同样的情况，说是上游的问题。

ok 刚才去看了 lede的luci已经合并分支了 应该重新编译一下就好了

MaxCrazy1101, 2024-03-10:
https://github.com/coolsnowwolf/luci/commit/aed573a82aef60bc2bf1c74a5488ee78bc56962c

closed 2024-03-10 by MaxCrazy1101